Universal-Debloater-Alliance/universal-android-debloater-next-generation #313: Add more realme system apps

kowith337, 2024-03-03:
### List your packages

## These seems to be an actual system apps, unlike previous issue that I've reported before.

**This mean I didn't removed that app yet** if appear in Removal section

### App Name: `92 Lock Screen for realme`
- Package name: `com.haokan.pictorial`
- List: OEM or Misc
- Removal: Recommended
- Description: 92 is pronounce in Chinese as 'Hao Kan' (it's company name), so it's China app, this not related to 'Lock Screen Magazine' and potentially inject ads in lock screen, I don't know.
- Hash: `162f76a86f105f8282da8d8301db91da77c850bdfa981c2e18427c6addc5110b`

### App Name: `AIUnit`
- Package name: `com.oplus.aiunit`
- List: OEM
- Removal: **Untested**
- Description: `AI Related`
- Hash: `39b2b4d11507c187837832dfc08c7b05a2d5489b105de7eb46d639e32f5ea27d`

### App Name: `AppHub`
- Package name: `com.applovin.array.apphub.vincere`
- List: Misc
- Removal: Recommended
- Description: `AppLovin` is Known as tracker company, potentially Adware!
- Hash: `bd6ff5d820f3774c03a6979fd69fb62ef59b1775aacff46bf149b89a6591089f`

### App Name: `Block & filter`
- Package name: `com.oplus.blacklistapp`
- List: OEM
- Removal: **Untested, probably Recommended**
- Description: Call blocking app that tied with device caller app, contains fuction to look caller location on Google Maps, potentially untrusted.
- Hash: `4ee3704564eab259dd44c663124bd438dd079229191a506666cd7fff2cb6c516`

### App Name: `Calculator`
- Package name: `com.coloros.calculator`
- List: OEM
- Removal: Recommended
- Description: Calculator app from ROM (Oppo/realme/Narzo where it use ColorOS), have internet access because it can convert exchange rate, but don't know they use for other things or not?
- Hash: `27859a408ea2d1e47329f7843b61abee673978cfb152a539bf68c50d94577f8c`

### App Name: `Camera`
- Package name: `com.oplus.camera`
- List: OEM
- Removal: **Untested, probably Advance**
- Description: Camera app from ROM (Oppo/realme/Narzo where it use ColorOS), due to related to Hardware, so remain untouched.
- Hash: `1c4239ef9daabe0225ca37f9f19302a1caf2b8e4ed0991c7e5377c0ba185828b`
\* too large to scan, so I use 7zip and parse SHA256 instead.

### App Name: `FIDO UAF1.0 Client`
- Package name: `com.fido.uafclient`
- List: Misc
- Removal: **Untested, probably Recommended**
- Description: Probably related to `FIDO` devices (Client), maybe safe to remove if you don't have any.
- Hash: `bf349b86002dccbc42eb4f5b1a30499b553b49d1127adbb2fcd33eff497164d9`

### App Name: `FIDO UAF1.0 ASM`
- Package name: `com.fido.asm`
- List: Misc
- Removal: **Untested, probably Recommended**
- Description: Probably related to `FIDO` devices (Framework?), maybe safe to remove if you don't have any.
- Hash: `30d8b0e5cf468fcfe69cb1bdf1e6954ed541b09baf4a269fbe3f4725ee93e357`

### App Name: `Fineasy`
- Package name: `com.fintech.life`
- List: Misc
- Removal: Recommended
- Description: China financial or payment app, note that it access locations, contacts, camera, mic and contain ads, Pithus report this as 'High Risk'
- Hash: `bd6c34559996ed1c3db893e4c21f3ca18a717c26984dba7d23df3bc92f252c3e`

### App Name: `Game Center`
- Package name: `com.nearme.gamecenter`
- List: OEM
- Removal: **Advance** (Cannot remove, only disable)
- Description: Act like QooApp that have game forum, news, etc. Contains tracker.
- Hash: `b265153cb1683c26502b642a01db3520b972f55a959cf67bcf49aee49d1a3972`
- \* too large to scan, so I use 7zip and parse SHA256 instead.

### App Name: `Number Origin`
- Package name: `com.oplus.phonenoareainquire`
- List: OEM
- Removal: **Untested**
- Description: Not yet uninstalled but due to it's related to phone calling function, need to check deeply...
- Hash: `e0bc788d0e1efca7b80e600008dfbbabdc36a3ceb9e08f035a360b2bdc652f89`

### App Name: `Screen recording`
- Package name: `com.oplus.screenrecorder`
- List: OEM
- Removal: **Advance**
- Description: Potentially lost screen recording function if uninstalled.
- Hash: `ba1754b719ef0eb0fb195385c492c52ea466ea8e14a8413b4934b2bc125c2d9e`

### App Name: `Theme Store`
- Package name: `com.oplus.themestore`
- List: OEM
- Removal: **Untested, probably Advance**
- Description: To change fonts, wallpapers, etc. of your device, note that you can download but you can't apply unless you have registered for/sign in with HeyTap account.
- Hash: `cd56832bc6ae99c8b01f67d1aa2368f3da8c5c8db776ebd2b0596bf0461e20df`
\* too large to scan, so I use 7zip and parse SHA256 instead.

### Acknowledgements

- [ ] I removed all those packages on my phone. If not, why? Explain in the issue.
- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-03-03:
Thanks for this detailed issue! I made a PR. Next time, please pay attention to the removal, as some seemed incorrect (I corrected them). The issue template explains when something should be `Recommended`.

Oh by the way, the hashes aren't necessary for next time. 😉

kowith337, 2024-03-04:
I knew the hash is optional, but because other contributors also put Pithus analysis, so I add the hash in case of investigating app behaviors too, even though some apps are too large to getting scan online.

closed 2024-03-05 by AnonymousWP

kowith337, 2024-03-14:
# Update app

### App Name: `FIDO UAF1.0 ASM`
Package name: `com.fido.asm`
List: Misc
Removal: Expert
Description: FIDO digital authenticator Framework, uninstalling this may cause bootloop, you can still uninstall other package that it's have 'client' named in.

### App Name: `Number Origin`
Package name: `com.oplus.phonenoareainquire`
List: OEM
Removal: Advance
Description: Display origin of phone number (estimate location based, such as county and region), this tied with stock phone app.

AnonymousWP, 2024-03-14:
> # Update app
>
> ### App Name: `FIDO UAF1.0 ASM`
>
> - Package name: `com.fido.asm`
> - List: Misc
> - Removal: Expert
> - Description: FIDO digital authenticator Framework, uninstalling this may cause [...]

Please open a new issue for this.