Universal-Debloater-Alliance/universal-android-debloater-next-generation #316: No recommended packages for Xiaomi Redmi Note 4 (MediaTek)

rei4877, 2024-03-04:
### Describe the bug

The device is Xiaomi Redmi Note 4 with MediaTek chip. The OS running is MIUI 10.

When the application detects the phone, the window shows no recommendation for the apps.
[image]

Although when the "Recommended" dropdown is changed to "All", it shows packages, although without any descriptions.
[image]

### Expected behavior

It should show all the recommended apps to debloat. The app shows the recommended apps on my other devices, namely POCO X3 Pro and Samsung Galaxy Note10 Lite.

### You have a solution?

Current solution to debloat said phone is using this https://github.com/kirthandev/MIUI-Debloater-official
Although it does not allow reinstallation of removed packages.

Also have tried it with this script https://github.com/ramalhovfc/miui10-11-debloater

### Provide logs

```
2024-03-03 08:08:42 DEBUG [src\core\update.rs:174] Checking for UAD-ng update
2024-03-03 08:09:57 INFO  [src\gui\views\list.rs:137] -----------------------------------------------------------------
2024-03-03 08:09:57 INFO  [src\gui\views\list.rs:138] ANDROID_SDK: 0 | DEVICE: fetching devices...
2024-03-03 08:09:57 INFO  [src\gui\views\list.rs:142] -----------------------------------------------------------------
2024-03-03 08:11:06 INFO  [src\gui\views\list.rs:137] -----------------------------------------------------------------
2024-03-03 08:11:06 INFO  [src\gui\views\list.rs:138] ANDROID_SDK: 23 | DEVICE: Xiaomi Redmi Note 4
2024-03-03 08:11:06 INFO  [src\gui\views\list.rs:142] -----------------------------------------------------------------
2024-03-03 08:11:24 DEBUG [src\core\update.rs:174] Checking for UAD update
2024-03-03 08:11:26 INFO  [src\gui\views\list.rs:125] -----------------------------------------------------------------
2024-03-03 08:11:26 INFO  [src\gui\views\list.rs:126] ANDROID_SDK: 23 | DEVICE: Xiaomi Redmi Note 4
2024-03-03 08:11:26 INFO  [src\gui\views\list.rs:130] -----------------------------------------------------------------
2024-03-03 08:13:32 DEBUG [src\core\update.rs:174] Checking for UAD-ng update
2024-03-03 08:13:35 INFO  [src\gui\views\list.rs:137] -----------------------------------------------------------------
2024-03-03 08:13:35 INFO  [src\gui\views\list.rs:138] ANDROID_SDK: 23 | DEVICE: Xiaomi Redmi Note 4
2024-03-03 08:13:35 INFO  [src\gui\views\list.rs:142] -----------------------------------------------------------------
2024-03-03 08:14:15 DEBUG [src\gui\mod.rs:271] UAD-ng update has been download!
2024-03-03 08:14:15 ERROR [src\gui\mod.rs:310] Failed to update UAD-ng!
2024-03-03 08:14:18 DEBUG [src\core\update.rs:174] Checking for UAD-ng update
2024-03-03 08:14:21 ERROR [src\core\utils.rs:82] Can't open the following URL: 
2024-03-03 08:14:30 INFO  [src\gui\views\list.rs:137] -----------------------------------------------------------------
2024-03-03 08:14:30 INFO  [src\gui\views\list.rs:138] ANDROID_SDK: 23 | DEVICE: Xiaomi Redmi Note 4
2024-03-03 08:14:30 INFO  [src\gui\views\list.rs:142] -----------------------------------------------------------------
2024-03-03 08:14:31 DEBUG [src\gui\mod.rs:271] UAD-ng update has been download!
2024-03-03 08:14:31 ERROR [src\gui\mod.rs:310] Failed to update UAD-ng!
2024-03-03 08:15:01 WARN  [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-hal-0.16.2\src\gles\egl.rs:226] No config found!
2024-03-03 08:15:01 WARN  [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-hal-0.16.2\src\gles\egl.rs:215] EGL says it can present to the window but not natively
2024-03-03 08:15:18 DEBUG [src\core\update.rs:174] Checking for UAD-ng update
2024-03-03 08:15:20 DEBUG [src\core\update.rs:174] Checking for UAD-ng update
2024-03-03 08:15:21 INFO  [src\gui\views\list.rs:136] -----------------------------------------------------------------
2024-03-03 08:15:21 INFO  [src\gui\views\list.rs:137] ANDROID_SDK: 23 | DEVICE: Xiaomi Redmi Note 4
2024-03-03 08:15:21 INFO  [src\gui\views\list.rs:141] -----------------------------------------------------------------
2024-03-03 08:15:31 INFO  [src\gui\views\list.rs:136] -----------------------------------------------------------------
2024-03-03 08:15:31 INFO  [src\gui\views\list.rs:137] ANDROID_SDK: 23 | DEVICE: Xiaomi Redmi Note 4
2024-03-03 08:15:31 INFO  [src\gui\views\list.rs:141] -----------------------------------------------------------------
2024-03-03 08:16:04 DEBUG [src\core\update.rs:174] Checking for UAD-ng update
2024-03-03 08:16:08 INFO  [src\gui\views\list.rs:136] -----------------------------------------------------------------
2024-03-03 08:16:08 INFO  [src\gui\views\list.rs:137] ANDROID_SDK: 23 | DEVICE: Xiaomi Redmi Note 4
2024-03-03 08:16:08 INFO  [src\gui\views\list.rs:141] -----------------------------------------------------------------
2024-03-03 08:16:10 INFO  [src\gui\views\list.rs:136] -----------------------------------------------------------------
2024-03-03 08:16:10 INFO  [src\gui\views\list.rs:137] ANDROID_SDK: 23 | DEVICE: Xiaomi Redmi Note 4
2024-03-03 08:16:10 INFO  [src\gui\views\list.rs:141] -----------------------------------------------------------------
2024-03-03 08:22:18 DEBUG [src\core\update.rs:174] Checking for UAD-ng update
2024-03-03 08:22:21 INFO  [src\gui\views\list.rs:136] -----------------------------------------------------------------
2024-03-03 08:22:21 INFO  [src\gui\views\list.rs:137] ANDROID_SDK: 23 | DEVICE: Xiaomi Redmi Note 4
2024-03-03 08:22:21 INFO  [src\gui\views\list.rs:141] -----------------------------------------------------------------
2024-03-03 08:22:35 DEBUG [src\core\update.rs:174] Checking for UAD-ng update
[... 153 lines cut ...]
 -> /system/bin/sh: cmd: not found
2024-03-03 13:28:14 INFO  [src\core\sync.rs:125] [Unlisted] cmd package install-existing --user 0 com.miui.translationservice
 -> /system/bin/sh: cmd: not found
2024-03-03 13:28:14 INFO  [src\core\sync.rs:125] [Unlisted] cmd package install-existing --user 0 com.miui.weather2
 -> /system/bin/sh: cmd: not found
2024-03-03 13:28:14 INFO  [src\core\sync.rs:125] [Unlisted] cmd package install-existing --user 0 com.xiaomi.joyose
 -> /system/bin/sh: cmd: not found
2024-03-03 13:28:14 INFO  [src\core\sync.rs:125] [Unlisted] cmd package install-existing --user 0 com.xiaomi.midrop
 -> /system/bin/sh: cmd: not found
2024-03-03 13:28:14 INFO  [src\core\sync.rs:125] [Unlisted] cmd package install-existing --user 0 com.xiaomi.mipicks
 -> /system/bin/sh: cmd: not found
2024-03-03 13:28:14 INFO  [src\core\sync.rs:125] [Unlisted] cmd package install-existing --user 0 com.xiaomi.payment
 -> /system/bin/sh: cmd: not found
2024-03-03 13:28:14 INFO  [src\core\sync.rs:125] [Unlisted] cmd package install-existing --user 0 com.xiaomi.scanner
 -> /system/bin/sh: cmd: not found
2024-03-03 13:28:26 INFO  [src\gui\views\list.rs:136] -----------------------------------------------------------------
2024-03-03 13:28:26 INFO  [src\gui\views\list.rs:137] ANDROID_SDK: 23 | DEVICE: Xiaomi Redmi Note 4
2024-03-03 13:28:26 INFO  [src\gui\views\list.rs:141] -----------------------------------------------------------------
```

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Frigyes06, 2024-03-04:
Please download the latest (1.0.2) release

rei4877, 2024-03-04:
On the original post I was using the latest release. I have re-downloaded and reused the latest release now, and there's still no packages recommendation.
[image]

Here is another screenshot where it includes when a package is clicked on (in this case, `android.autoinstallconfig.Xiaomi.nikel`)
[image]

AnonymousWP, 2024-03-04:
Because that package isn't even on our list: 
[image]

rei4877, 2024-03-04:
Will posting all the packages name within this phone help?

AnonymousWP, 2024-03-05:
> Will posting all the packages name within this phone help?

Yes, it should, but only document the packages that are not in the list. Choose this issue template and please read it thoroughly: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/new?assignees=&labels=package%3A%3Aaddition&projects=&template=3-add-new-packages.yml&title=pkg%28scope%29%3A+

AnonymousWP, 2024-03-14:
Closing this issue as completed due to instructing the user.

closed 2024-03-14 by AnonymousWP