Universal-Debloater-Alliance/universal-android-debloater-next-generation #324: bug(scope): com.oneplus.camera.service - bootloop on Android 14 ROM

Mazoch, 2024-03-10:
### Describe the problem

Android 14: Nameless-AOSP_kebab-14.0-20240229-1518
firmware: fw_Oneplus8t_EU_OOS14_H19_30
app: com.oneplus.camera.service

Currently, at least in my case, I can not remove this package because the phone falls into bootloop. On the official OOS 14 system everything works fine, so it's a matter of this ROM

### You have a solution?

Add information that warns that bootloop may occur on some ROMs with Android 14.

### Provide logs

<pre><code>2024-03-10 14:39:03 DEBUG [src\core\update.rs:174] Checking for UAD-ng update
2024-03-10 14:39:03 INFO  [src\gui\views\list.rs:136] -----------------------------------------------------------------
2024-03-10 14:39:03 INFO  [src\gui\views\list.rs:137] ANDROID_SDK: 34 | DEVICE: OnePlus KB2003
2024-03-10 14:39:03 INFO  [src\gui\views\list.rs:141] -----------------------------------------------------------------
2024-03-10 14:39:09 INFO  [src\core\sync.rs:125] [Recommended] pm uninstall --user 0 com.oneplus.camera.service -> Success
2024-03-10 14:39:16 INFO  [src\core\sync.rs:125] [Recommended] cmd package install-existing --user 0 com.oneplus.camera.service -> Package com.oneplus.camera.service installed for user: 0</pre></code>

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-03-10:
Hi fellow OnePlus user,

For some reason it's no longer present it seems, while I've certainly seen that package on Android 13.1. Do you think this should remain `Recommended` with a notice about custom ROMs, or `Unsafe` but mention this is recommended to remove when using OxygenOS? I believe we should go for the former and put "WARNING" or "NOTE", but keep it `Recommended` as removal.

Mazoch, 2024-03-13:
Hi, thank you for your reply. I think it is enough to add a warning. The application is already noticeable at a glance, because it is something from one plus, so in case of a bootloop such information will help to locate the faulty package. Nonetheless, it is a useless application and perhaps in most people it will remove itself correctly. Adding a warning is probably a good compromise.

Mazoch, 2024-03-14:
@AnonymousWP Please add that this only applies to Android 14. On Android 13 or below I don't notice any issues. 

`WARNING: may cause a bootloop when using a custom ROM on Android 14 (in this case, Nameless AOSP). Does not affect users running OxygenOS.`

Thank you for your work I really appreciate that

AnonymousWP, 2024-03-14:
Done, thanks. 🙂 And please keep this issue opened; once my PR gets merged, this issue is closed automatically.

closed 2024-03-14 by AnonymousWP