Universal-Debloater-Alliance/universal-android-debloater-next-generation #326: feat(scope): not really request, but package upgrade to iced-0.12

edgy-b, 2024-03-10:
### Describe the feature you want

Hi.

I performed iced upgrade to 0.12. It's working, but with workaround, let me explain.
https://github.com/edgy-b/universal-android-debloater-next-generation/tree/feat/ICED-12
Could you create a copy of master branch and I'll merge my branch into it for testing and further improvements.

In v0.10 `pick_list` function differs from v0.12
https://github.com/iced-rs/iced/blob/0.10/widget/src/helpers.rs
https://github.com/iced-rs/iced/blob/0.12/widget/src/helpers.rs#L286

My workaround consists of following changes:
 - `src/gui/widgets/navigation_menu.rs` on `ln: 91` add `.borrow()` or `.clone()` to `device_list`
 - `src/gui/widgets/modal.rs` lines `167 - 170` remove `mut` from `child` and `node` and also `child.clone().align.....`

Reverse these workarounds and there are compile errors
```
error[E0277]: the trait bound `&Vec<Phone>: Borrow<[_]>` is not satisfied
   --> src/gui/widgets/navigation_menu.rs:90:23
    |
90  |             pick_list(device_list, Some(phone), Message::DeviceSelected,),
    |             --------- ^^^^^^^^^^^ the trait `Borrow<[_]>` is not implemented for `&Vec<Phone>`
    |             |
    |             required by a bound introduced by this call
    |
    = help: the following other types implement trait `Borrow<Borrowed>`:
              <Vec<u8> as Borrow<winnow::stream::Bytes>>
              <Vec<u8> as Borrow<winnow::stream::BStr>>
              <Vec<T, A> as Borrow<[T]>>
note: required by a bound in `pick_list`
   --> /home/edga/.cargo/registry/src/index.crates.io-6f17d22bba15001f/iced_widget-0.12.3/src/helpers.rs:275:8
    |
268 | pub fn pick_list<'a, T, L, V, Message, Theme, Renderer>(
    |        --------- required by a bound in this function
...
275 |     L: Borrow<[T]> + 'a,
    |        ^^^^^^^^^^^ required by this bound in `pick_list`

error[E0277]: the trait bound `&Vec<Phone>: Borrow<[Phone]>` is not satisfied
  --> src/gui/widgets/navigation_menu.rs:90:13
   |
90 |             pick_list(device_list, Some(phone), Message::DeviceSelected,),
   |             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `Borrow<[Phone]>` is not implemented for `&Vec<Phone>`
   |
   = help: the following other types implement trait `Borrow<Borrowed>`:
             <Vec<u8> as Borrow<winnow::stream::Bytes>>
             <Vec<u8> as Borrow<winnow::stream::BStr>>
             <Vec<T, A> as Borrow<[T]>>
note: required by a bound in `iced::widget::PickList`
  --> /home/edga/.cargo/registry/src/index.crates.io-6f17d22bba15001f/iced_widget-0.12.3/src/pick_list.rs:36:8
   |
26 | pub struct PickList<
   |            -------- required by a bound in this struct
...
36 |     L: Borrow<[T]> + 'a,
   |        ^^^^^^^^^^^ required by this bound in `PickList`

error[E0277]: the trait bound `&Vec<Phone>: Borrow<[Phone]>` is not satisfied
  --> src/gui/widgets/navigation_menu.rs:87:24
   |
87 |           Some(phone) => row![
   |  ________________________^
88 | |             reboot_btn,
89 | |             apps_refresh_tooltip,
90 | |             pick_list(device_list, Some(phone), Message::DeviceSelected,),
...  |
96 | |             settings_btn,
97 | |         ]
   | |_________^ the trait `Borrow<[Phone]>` is not implemented for `&Vec<Phone>`
   |
   = help: the following other types implement trait `Borrow<Borrowed>`:
             <Vec<u8> as Borrow<winnow::stream::Bytes>>
             <Vec<u8> as Borrow<winnow::stream::BStr>>
             <Vec<T, A> as Borrow<[T]>>
   = note: required for `iced::advanced::iced_graphics::iced_core::Element<'_, gui::Message, _, _>` to implement `From<iced::widget::PickList<'_, Phone, &Vec<Phone>, Phone, gui::Message, _, _>>`
   = note: this error originates in the macro `row` (in Nightly builds, run with -Z macro-backtrace for more info)
```
Whick is a bit too much to handle for me, yet, learned rust in 2 hours, just to help this project, but this issues giving me headache. So any advice or someone could take a look what are necessary changes for it to work without `clone` or `borrow` implementation.
Any rust gurus here?
Don't want to go clone or borrow way, https://rust-unofficial.github.io/patterns/anti_patterns/borrow_clone.html

### Acknowledgements

- [X] This issue is not a duplicate of an existing feature request.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-03-10:
@Universal-Debloater-Alliance/developers Please see this.

edgy-b, 2024-03-10:
It's giving me 404 page

edgy-b, 2024-03-12:
solved

closed 2024-03-18 by AnonymousWP