Universal-Debloater-Alliance/universal-android-debloater-next-generation #328: bug(scope): Unable to run the program

scholablade, 2024-03-10:
### Describe the bug

Once I run the program, it crashes, I have built it from source.

### Expected behavior

Running.

### Provide logs

```
2024-03-10 20:57:16 ERROR [src/core/config.rs:77] Invalid config file: `TOML parse error at line 1, column 1
  |
1 | theme = "Lupin"
  | ^^^^^^^^^^^^^^^
missing field `general`
`
2024-03-10 20:57:16 ERROR [src/core/config.rs:81] Restoring default config file
2024-03-10 20:57:23 ERROR [/home/rextheman/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-0.16.3/src/backend/direct.rs:594] Error in Adapter::request_device: Limit 'max_compute_workgroups_per_dimension' value 65535 is better than allowed 0
2024-03-10 20:57:23 ERROR [/home/rextheman/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-0.16.3/src/backend/direct.rs:594] Error in Adapter::request_device: Limit 'max_compute_workgroups_per_dimension' value 65535 is better than allowed 0
```

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Frigyes06, 2024-03-12:
What OS?

drrescher, 2024-03-15:
I guess Windows - i face the same issue. The only version i was able to run was the opengl variant of version 0.6.2. But there is no longer an opengl version (as described in the wiki) from version 1.0 on..

But tbh i havent read the entire wiki yet, so didnt dare to open a ticket myself ..

edgy-b, 2024-03-16:
For windows go to `C:\Users\Alicewonderland\AppData\Roaming` find uad folder and delete config.toml file

edgy-b, 2024-03-18:
> hi @edgy-b that didn't help me. I also can't run uad unless it's the opengl version. For the record this is the contents of config.toml
> 
> ```
> [general]
> theme = ""
> expert_mode = false
> ```
> 
> and the error
> 
> ```
> 2024-03-18 07:13:20 ERROR [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-hal-0.16.2\src\auxil\dxgi\result.rs:25] Command allocator creation failed: 0x887A0005
> 2024-03-18 07:13:20 ERROR [C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-0.16.3\src\backend\direct.rs:3018] Handling wgpu errors as fatal by default
> thread 'main' panicked at C:\Users\runneradmin\.cargo\registry\src\index.crates.io-6f17d22bba15001f\wgpu-0.16.3\src\backend\direct.rs:3019:5:
> wgpu error: Out of Memory
> 
> note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
> ```

It's seems like wgpu error, there are some isssues created for wgpu
Take a look https://github.com/gfx-rs/wgpu/issues
Meanwhile if its running fine use opengl

jflambert, 2024-03-18:
@edgy-b My current issue is with v1.0.2.

if I run the previous opengl binary v0.6.2, it opens, but stays stuck at pulling list from github.

I came here following a link from the https://github.com/0x192/universal-android-debloater project, which appears to be dead.

AnonymousWP, 2024-03-18:
Probably related to https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/pull/119.

edgy-b, 2024-03-19:
> @edgy-b My current issue is with v1.0.2.
> 
> if I run the previous opengl binary v0.6.2, it opens, but stays stuck at pulling list from github.
> 
> I came here following a link from the https://github.com/0x192/universal-android-debloater project, which appears to be dead. Their last version v0.5.1 worked fine, except it is obsolete of course.

We upgraded iced version, there were some fixes to wgpu.
We hope to release 1.0.3 soon, 2 issues to fix.

scholablade, 2024-04-11:
Am closing this issue, thanks for the help!

closed 2024-04-11 by scholablade