Universal-Debloater-Alliance/universal-android-debloater-next-generation #330: mipad (kitkat) home and recent buttons stopped working after removing recommended + 2 other packages

bheeshmpita, 2024-03-12:
### Describe the problem

restarting after debloating the device rendered those 2 buttons not working. Tried home and recent actions from the touch assistant/quick ball button but results are same. Those (home and recent) dont work.
Restoring the debloated packages and restarting the device turns the functionality to normal.
How can restore the functionality with debloating and how can i find out which disabled package is causing the issue?

### Provide logs

[UAD_20240312.log](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/files/14575729/UAD_20240312.log)

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Mazoch, 2024-03-13:
If you want to check which packages are causing problems then I recommend the method of uninstalling 5-10 packages at a time and testing each time if the buttons work. This will help you narrow down your search area. I don't think it can be done otherwise.

The log file doesn't work, I don't know if it's github's fault or something you did wrong.

bheeshmpita, 2024-03-13:
i have uploaded the log file here.
[https://hastebin.com/share/icozolates.yaml](url)
is this link working?

AnonymousWP, 2024-03-13:
> The log file doesn't work, I don't know if it's github's fault or something you did wrong.

Works fine here. It prompts me to download the log file to my computer.

closed 2024-03-28 by bheeshmpita

AnonymousWP, 2024-03-30:
> the package that was breaking the system was `com.android.keyguard`

Thanks, I'll improve the documentation for this.