Universal-Debloater-Alliance/universal-android-debloater-next-generation #346: bug(scope): MiRemote is not listed

wooque, 2024-03-17:
### Describe the bug

MiRemote is not listed, even when choosing "All" -> "All packages". Package name is **com.duokan.phone.remotecontroller**

### Expected behavior

Show com.duokan.phone.remotecontroller under some of the lists, Advanced maybe?

### Provide logs

[UAD_20240317.log](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/files/14627214/UAD_20240317.log)

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

xe1st, 2024-03-17:
MiRemote is removable without adb; that's why it isn't in the list.

wooque, 2024-03-17:
@xe1st oh, yes, you are right :) I expected all bloatware to be shown, but yes I can remove it without UDA.

closed 2024-03-17 by wooque