Universal-Debloater-Alliance/universal-android-debloater-next-generation #347: bug(scope): UAD won't open on my laptop

llyqd, 2024-03-17:
### Describe the bug

UAD on my laptop wont let me open.

### Expected behavior

it won't open no matter how I make an alternative solution.

### You have a solution?

idk.

### Provide logs

.

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

edgy-b, 2024-03-17:
ADB installed?

AnonymousWP, 2024-03-17:
If you're putting so little effort in filling in the template, then why should we put effort into helping you? You haven't posted any logs, so obviously we can't help you properly.

Abigail-17, 2024-03-17:
Please at least say what OS you are running, what version, and which package you downloaded from the Releases section

llyqd, 2024-03-20:
> ADB installed?
yes.

llyqd, 2024-03-20:
> If you're putting so little effort in filling in the template, then why should we put effort into helping you? You haven't posted any logs, so obviously we can't help you properly.

apologies 😅

AnonymousWP, 2024-04-12:
@llyqd Are you going to follow up on this?

Frigyes06, 2024-05-04:
Closing due to inactivity

closed 2024-05-04 by Frigyes06