Universal-Debloater-Alliance/universal-android-debloater-next-generation #350: Remove com.sec.android.diagmonagent from recommended packages

Lost-Entrepreneur439, 2024-03-17:
### List your packages

com.sec.android.diagmonagent says in the app description that it spams the crash log, overheats the phone, and drains battery. There is no reason this should be a recommended package.

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-03-17:
Thanks for reporting, but next time please stick to the issue template. I'll get this fixed.

AnonymousWP, 2024-03-17:
Hmm, this is a difficult one; it sends diagnostics to Samsung, which makes it recommended, but then making the device overheat is a serious problem too. I'm thinking of making it `Advanced` *and* put a warning in the description.

Do you know if this problem also occurs when uninstalling the package instead of disabling?

Lost-Entrepreneur439, 2024-03-17:
I'm unsure, it was my friend who noticed this on her S22 Ultra and she didn't want to try uninstalling it, I don't have any modern Samsung devices personally.

AnonymousWP, 2024-03-17:
Okay, and would you be okay with the idea I proposed?

Lost-Entrepreneur439, 2024-03-17:
Yes

closed 2024-03-17 by AnonymousWP

d0ob, 2026-03-12:
Hi there ! 
Sorry to post on a closed issue, I have a question related :

From Canta, which is apparently fetching informations from here, that bloat is still recommended to remove ; Has it been proven safe to remove since ? Or is it old infos ?

Thanks for all the work, in any case, treasur of information !