Universal-Debloater-Alliance/universal-android-debloater-next-generation #353: docs(readme/contributing/wiki): streamline documentation

AnonymousWP, 2024-03-17:
### Describe the documentation you want to change

At the moment, our documentation is quite split. For example, we want to keep the README minimal and should serve as a summary for this repository, like [how it should be](https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/about-readmes#wikis). However, we also have a `CONTRIBUTING.md` file and a Wiki. We need to draw a line, for instance:

- Do we drop the Wiki entirely? Rigorous change though.
- Do we move [the Home page](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki) to the README?
- Do we merge [the "How to contribute"](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/How-to-contribute) with our `CONTRIBUTING.md`?

### Acknowledgements

- [X] This issue is not a duplicate of an existing documentation change request.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

edgy-b, 2024-03-18:
There is aboslutely no need to rewrite wiki.
We can add section
## Quick startup
 - install [adb](https://www.xda-developers.com/install-adb-windows-macos-linux/)
 - use the application

AnonymousWP, 2024-03-24:
Thanks for your input.

But I do not want to put a quick start in the README, as it's then hard to draw a line, sort of. The "Getting started" page is at the bottom, and I can't fix that, unless I customise the pages which take a bit of work.

> maybe change wording For documentation. I believe in users eyes documentation is for dev's, gone is the age of terminal where user inserts floppy disc and really has to read documentation how to use software.

This I can understand. I have [fixed that](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation?tab=readme-ov-file#documentation) a short while ago, what do you think? Would it be more clear now? Because the README is short, users would easily see the Documentation header and thus how to get started.