Universal-Debloater-Alliance/universal-android-debloater-next-generation #355: bug(linux, startup): The program hangs when I click anywhere in the program window

Nord1cWarr1or, 2024-03-18:
### Describe the bug

After starting, if I click anywhere within the program, it hangs, no matter if the device is connected or not.

[image]

OS: Linux-x64, Arch-based distro, KDE Plasma 6.0.2, Wayland.
Version: uad-ng-linux v1.0.2

### Expected behavior

The program works

### Provide logs

[UAD_20240318.log](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/files/14636129/UAD_20240318.log)

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Nord1cWarr1or, 2024-03-18:
> As you can see in your logs, ADB is not installed. Therefore, it doesn't work. Please read https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/Getting-started. Let us know if this worked.

No, it's not. Please note the timestamp of the messages.
The posts from 15:16:26 onwards are already with adb installed.

I have installed the `android-tools-34.0.4-4` package

Nord1cWarr1or, 2024-03-18:
I have now discovered that the program behavior I described is only relevant for the Wayland display protocol. When I use X11, the program works fine.

AnonymousWP, 2024-03-18:
So if I'm right, this seems to be related to https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/53.

Nord1cWarr1or, 2024-03-18:
> So if I'm right, this seems to be related to #53.

Yeah, it looks like the app just doesn't work on Wayland. Sad :(
That guy is using intel/amd graphics and I have nvidia and have a similar problem.

AnonymousWP, 2024-03-18:
In that case I'll close this issue as duplicate. In case you found a solution or can come up with other findings, please share it in that issue.

closed 2024-03-18 by AnonymousWP