Universal-Debloater-Alliance/universal-android-debloater-next-generation #358: pkg(com.xiaomi.midrop): Xiaomi ShareMe

vitoramaral10, 2024-03-18:
### List your packages

- Package name: com.xiaomi.midrop
- List: OEM
- Removal: Recommended
- Description: is used to share files

### Acknowledgements

- [X] I removed all those packages on my phone. If not, why? Explain in the issue.
- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-03-18:
Thanks for your issue. However, as the issue template said: if you have multiple packages, they can be combined into 1 issue. 😉

closed 2024-03-18 by AnonymousWP

vitoramaral10, 2024-03-18:
I've checked here and it's on the list, but it doesn't appear to be removed, is there a problem with that?