Universal-Debloater-Alliance/universal-android-debloater-next-generation #361: bug(scope): App in the list (json) are installed on the cell phone but do not appear in UAD

vitoramaral10, 2024-03-18:
### Describe the bug

I ended up opening 3 issues because I thought they weren't on the list (my mistake for not seeing the json, sorry), but they don't appear in the UAD for removal from the cell phone

https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/357
https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/358
https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/359

Mi device is: Xiaomi Redmi 10 5G
Android 14 with HyperOS

### Expected behavior

The expectation was that they would show up to uninstall

### Provide logs

2024-03-18 15:17:35 DEBUG [src\core\update.rs:174] Checking for UAD-ng update
2024-03-18 15:17:35 INFO  [src\gui\views\list.rs:136] -----------------------------------------------------------------
2024-03-18 15:17:35 INFO  [src\gui\views\list.rs:137] ANDROID_SDK: 34 | DEVICE: Redmi 22041219NY
2024-03-18 15:17:35 INFO  [src\gui\views\list.rs:141] -----------------------------------------------------------------

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-03-18:
So you're saying they don't appear at all or do you mean that they can't be uninstalled?

vitoramaral10, 2024-03-18:
it simply doesn't appear, even setting all filters to all

[image]

[image]

[image]

AnonymousWP, 2024-03-18:
I see, but this is a duplicate of https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/82. You checked the checkbox "This issue is not a duplicate of an existing bug report." but apparently you didn't check. 😉 We can continue there.

closed 2024-03-18 by AnonymousWP