Universal-Debloater-Alliance/universal-android-debloater-next-generation #370: bug(scope):

discomposed8890, 2024-03-23:
### Describe the bug

Program stuck in "Downloading latest UAD-ng lists from GitHub. Please wait..." 

### Expected behavior

Downloading lists only took 1-2 seconds after opening the program

### You have a solution?

No solution found yet 

### Provide logs

[UAD_20240323.log](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/files/14730401/UAD_20240323.log)

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

anime-shed, 2024-03-23:
Same here:
[image]

anime-shed, 2024-03-23:
Is there an option to skip downloading and use the list that is already there??

discomposed8890, 2024-03-23:
> Is there an option to skip downloading and use the list that is already there??

> Same here: [...]

It worked for me yesterday and today it stopped working

Knud3, 2024-03-23:
I was also coming to open issue about this. Its been down for about 11+ hours now.

I tried to delete everything about it from AppData and start with fresh "installation".

Starting app with settings and everything deleted:
```
2024-03-23 01:58:56 ERROR [src\core\config.rs:79] Failed to read config file: `The system cannot find the file specified. (os error 2)`
2024-03-23 01:58:56 ERROR [src\core\config.rs:81] Restoring default config file
2024-03-23 01:58:57 DEBUG [src\core\update.rs:174] Checking for UAD-ng update
2024-03-23 01:58:57 INFO  [src\gui\views\list.rs:136] -----------------------------------------------------------------
2024-03-23 01:58:57 INFO  [src\gui\views\list.rs:137] ANDROID_SDK: 34 | DEVICE: samsung SM-S928B
2024-03-23 01:58:57 INFO  [src\gui\views\list.rs:141] -----------------------------------------------------------------
```

After pressing "No internet?":
```
2024-03-23 03:17:02 INFO  [src\gui\views\list.rs:136] -----------------------------------------------------------------
2024-03-23 03:17:02 INFO  [src\gui\views\list.rs:137] ANDROID_SDK: 34 | DEVICE: samsung SM-S928B
2024-03-23 03:17:02 INFO  [src\gui\views\list.rs:141] -----------------------------------------------------------------
2024-03-23 03:17:02 WARN  [src\core\uad_lists.rs:208] Could not load remote debloat list
```

Even after that it got downloaded `uad_lists.json`, but got stuck to download screen.

Knud3, 2024-03-23:
@AnonymousWP check this for line 35752 https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/commit/c757f1a5067833a485d5de0dc5111dd60c5df5b4#diff-7b60c2b0f463a050d9284836dd992185ee8e09808f2b8432fccf85337e223ba0R35752

Knud3, 2024-03-23:
So, workaround is to open `%userprofile%\AppData\Local\uad\uad_lists.json` and delete everything from line `35751` and before last `}`. After that press `No internet?`

closed 2024-03-23 by AnonymousWP

AnonymousWP, 2024-03-24:
Hi, apologies for this issue. I just published a statement regarding this: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/discussions/375.

PR #371 (2024-03-23): bug(uad-list): json parsing error
 - fixed json file
 - added more verbose explanation where update failed in update.rs file

resolves #370