Universal-Debloater-Alliance/universal-android-debloater-next-generation #383: bug(scope): 1.0.3 core dump

yochananmarqos, 2024-03-26:
### Describe the bug

AUR package maintainer here. 1.0.3 fails to start while 1.0.2 launches fine. Log is attached.

### Expected behavior

N/A

### Provide logs

[UAD_20240326.log](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/files/14761848/UAD_20240326.log)

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

edgy-b, 2024-03-26:
> ### Describe the bug
> 
> AUR package maintainer here. 1.0.3 fails to start while 1.0.2 launches fine. Log is attached.
> 
> EDIT: This only happens on my NVIDIA machine running a GNOME Xorg session . I cannot reproduce on my AMD machine running a GNOME Wayland session.
> 
> ### Expected behavior
> 
> N/A
> 
> ### You have a solution?
> 
> _No response_
> 
> ### Provide logs
> 
> [UAD_20240326.log](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/files/14761848/UAD_20240326.log)
> 
> 
> ### Acknowledgements
> 
> - [X] This issue is not a duplicate of an existing bug report.
> - [X] I have chosen an appropriate title.
> - [X] All requested information has been provided properly.

This seems to be related to https://github.com/rust-windowing/winit/pull/3574

yochananmarqos, 2024-03-26:
@edgy-b I honestly have no idea how a two year old issue is related. In this case, the GUI never opens. 

Also, please don't quote the issue in your reply.

edgy-b, 2024-03-26:
> @edgy-b I honestly have no idea how a two year old issue is related. In this case, the GUI never opens. 
> 
> Also, please don't quote the issue in your reply.

Seesh didnt noticed that its so old

Frigyes06, 2024-03-30:
@phoropter Please open a separate issue

@yochananmarqos could you try updating the lockfile to latest and build to see if it got fixed in the meantime?

I opened an issue at winit because they were modifying a lot, but iced could also be at fault.

yochananmarqos, 2024-03-31:
@Frigyes06 The `Cargo.lock` hasn't been touched since the 1.0.3 tag.

Frigyes06, 2024-03-31:
@yochananmarqos Can you run uad with xtrace and upload the logs?

yochananmarqos, 2024-03-31:
I'm guessing you mean xtrace as in the Debian package? All I have is `/usr/bin/xtrace` as a part of `glibc`.

Frigyes06, 2024-03-31:
I love it when people name things the exact same. I think it'd be best if you talked directly to the winit guys over at https://github.com/rust-windowing/winit/issues/3616

AnonymousWP, 2024-09-15:
Closing this as it's stale. Try v1.1.0.

yochananmarqos, 2024-09-15:
@AnonymousWP 

With 1.10:

```
2024-09-15 13:23:41 ERROR [src/core/config.rs:92] Invalid config file: `TOML parse error at line 1, column 1
  |
1 | [general]
  | ^^^^^^^^^
missing field `backup_folder`
`
2024-09-15 13:23:41 ERROR [src/core/config.rs:96] Restoring default config file
2024-09-15 13:23:42 WARN  [/build/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/gles/egl.rs:268] No config found!
2024-09-15 13:23:42 WARN  [/build/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/gles/egl.rs:257] EGL says it can present to the window but not natively
2024-09-15 13:23:43 ERROR [/build/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/mod.rs:713] X11 error: XError {
    description: "BadMatch (invalid parameter attributes)",
    error_code: 8,
    request_code: 149,
    minor_code: 4,
}
2024-09-15 13:23:43 ERROR [/build/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/mod.rs:713] X11 error: XError {
    description: "BadMatch (invalid parameter attributes)",
    error_code: 8,
    request_code: 149,
    minor_code: 4,
}
2024-09-15 13:23:43 ERROR [/build/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/mod.rs:713] X11 error: XError {
    description: "BadMatch (invalid parameter attributes)",
    error_code: 8,
    request_code: 149,
    minor_code: 4,
}
thread 'main' panicked at /build/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/x11/window.rs:1208:18:
Failed to call XMapRaised: XError { description: "BadMatch (invalid parameter attributes)", error_code: 8, request_code: 149, minor_code: 4 }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
thread 'main' panicked at /build/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/x11/window.rs:601:34:
called `Result::unwrap()` on an `Err` value: PoisonError { .. }
stack backtrace:
   0:     0x55cdf3b77cef - <unknown>
   1:     0x55cdf37d7d5b - <unknown>
   2:     0x55cdf3b4442e - <unknown>
   3:     0x55cdf3b7c43e - <unknown>
   4:     0x55cdf3b7d4ad - <unknown>
   5:     0x55cdf3b7cde5 - <unknown>
   6:     0x55cdf3b7cd49 - <unknown>
   7:     0x55cdf3b7cd34 - <unknown>
   8:     0x55cdf374daf2 - <unknown>
   9:     0x55cdf374dff5 - <unknown>
  10:     0x55cdf3f70c07 - <unknown>
  11:     0x55cdf3cee59c - <unknown>
  12:     0x55cdf3ceea12 - <unknown>
  13:     0x55cdf3cf8fcf - <unknown>
  14:     0x55cdf3ce7e0d - <unknown>
  15:     0x55cdf3cd1142 - <unknown>
  16:     0x55cdf3d16e0f - <unknown>
  17:     0x55cdf3cd0170 - <unknown>
  18:     0x7f37cf9f6e08 - <unknown>
  19:     0x7f37cf9f6ecc - __libc_start_main
  20:     0x55cdf377e705 - <unknown>
  21:                0x0 - <unknown>
thread 'main' panicked at library/core/src/panicking.rs:229:5:
panic in a destructor during cleanup
thread caused non-unwinding panic. aborting.
zsh: IOT instruction (core dumped)  uad-ng
```

AnonymousWP, 2024-09-16:
1.1.0?

edgy-b, 2024-09-16:
@yochananmarqos 
Try deleting config file from ~/username/.config/uad/config.toml and open Uad again

yochananmarqos, 2024-09-16:
@edgy-b No difference.

edgy-b, 2024-09-16:
> @edgy-b No difference. 

@yochananmarqos 
How doea your config.toml file content looks like?

yochananmarqos, 2024-09-16:
@edgy-b 

```
[general]
theme = "Auto (follow system theme)"
expert_mode = false
backup_folder = "/home/yochanan/.cache/uad/backups"
```

edgy-b, 2024-09-20:
@yochananmarqos what if you try running `WEBKIT_DISABLE_DMABUF_RENDERER=1 cargo run`

yochananmarqos, 2024-09-20:
@edgy-b 

```
❯ WEBKIT_DISABLE_DMABUF_RENDERER=1 uad-ng
2024-09-20 14:11:03 WARN  [/build/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/gles/egl.rs:268] No config found!
2024-09-20 14:11:03 WARN  [/build/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/gles/egl.rs:257] EGL says it can present to the window but not natively
2024-09-20 14:11:03 ERROR [/build/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/mod.rs:713] X11 error: XError {
    description: "BadMatch (invalid parameter attributes)",
    error_code: 8,
    request_code: 149,
    minor_code: 4,
}
2024-09-20 14:11:03 ERROR [/build/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/mod.rs:713] X11 error: XError {
    description: "BadMatch (invalid parameter attributes)",
    error_code: 8,
    request_code: 149,
    minor_code: 4,
}
2024-09-20 14:11:03 ERROR [/build/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/mod.rs:713] X11 error: XError {
    description: "BadMatch (invalid parameter attributes)",
    error_code: 8,
    request_code: 149,
    minor_code: 4,
}
thread 'main' panicked at /build/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/x11/window.rs:1208:18:
Failed to call XMapRaised: XError { description: "BadMatch (invalid parameter attributes)", error_code: 8, request_code: 149, minor_code: 4 }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
thread 'main' panicked at /build/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/x11/window.rs:601:34:
called `Result::unwrap()` on an `Err` value: PoisonError { .. }
stack backtrace:
   0:     0x57ae6770acef - <unknown>
   1:     0x57ae6736ad5b - <unknown>
   2:     0x57ae676d742e - <unknown>
   3:     0x57ae6770f43e - <unknown>
   4:     0x57ae677104ad - <unknown>
   5:     0x57ae6770fde5 - <unknown>
   6:     0x57ae6770fd49 - <unknown>
   7:     0x57ae6770fd34 - <unknown>
   8:     0x57ae672e0af2 - <unknown>
   9:     0x57ae672e0ff5 - <unknown>
  10:     0x57ae67b03c07 - <unknown>
  11:     0x57ae6788159c - <unknown>
  12:     0x57ae67881a12 - <unknown>
  13:     0x57ae6788bfcf - <unknown>
  14:     0x57ae6787ae0d - <unknown>
  15:     0x57ae67864142 - <unknown>
  16:     0x57ae678a9e0f - <unknown>
  17:     0x57ae67863170 - <unknown>
  18:     0x7a5bdfa5de08 - <unknown>
  19:     0x7a5bdfa5decc - __libc_start_main
  20:     0x57ae67311705 - <unknown>
  21:                0x0 - <unknown>
thread 'main' panicked at library/core/src/panicking.rs:229:5:
panic in a destructor during cleanup
thread caused non-unwinding panic. aborting.
zsh: IOT instruction (core dumped)  WEBKIT_DISABLE_DMABUF_RENDERER=1 uad-ng
```

Rudxain, 2025-01-09:
> ```toml
> theme = "Auto (follow system theme)"
> ```

I suspected this would happen if we didn't `impl ToString` in addition to `impl Display` 😅. It should serialize to simply `"Auto"`. Silly me 🤦 (#531)

AnonymousWP, 2025-01-13:
@yochananmarqos @phoropter  Sorry that it took this long. Could you try with one of these artifacts? https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/actions/runs/12749084042

yochananmarqos, 2025-01-14:
@AnonymousWP No worries. 

Thing is, I'm now using Wayland. I've lost interest in pursuing this issue. Feel free to close it if you feel the same.

closed 2025-02-18 by AnonymousWP