Universal-Debloater-Alliance/universal-android-debloater-next-generation #394: bug(scope): It is not possible to run this application on a Windows 11 laptop

Satorllous, 2024-03-29:
### Describe the bug

I run the program, it closes immediately after opening, there are no errors

### Expected behavior

fix

### You have a solution?

No response

### Provide logs

no

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-03-29:
This is strange, works fine here on multiple Windows 11 machines. We do need logs though, so post them here.

Satorllous, 2024-03-30:
Log Name:      Application
Source:        Application Error
Date:          30.03.2024 12:19:35
Event ID:      1000
Task Category: Application Crashing Events
Level:         Ошибка
Keywords:      
User:          DESKTOP-7DH039P\Алексей
Computer:      DESKTOP-7DH039P
Description:
Faulting application name: uad-ng-windows.exe, version: 0.0.0.0, time stamp: 0x660187c7
Faulting module name: nvoglv64.dll_unloaded, version: 31.0.15.5186, time stamp: 0x65f0dfad
Exception code: 0xc0000005
Fault offset: 0x0000000000fbc5cc
Faulting process ID: 0x0x30E4
Faulting application start time: 0x0x1DA82729C4CDDD7
Faulting application path: C:\Users\Алексей\Desktop\uad-ng-windows.exe
Faulting module path: nvoglv64.dll
Report ID: 7f014af8-f7e3-4e6b-b7ea-f6a928f3d6c4
Faulting package full name: 
Faulting package-relative application ID: 
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" Guid="{a0e9b465-b939-57d7-b27d-95d8e925ff57}" />
    <EventID>1000</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>100</Task>
    <Opcode>0</Opcode>
    <Keywords>0x8000000000000000</Keywords>
    <TimeCreated SystemTime="2024-03-30T07:19:35.4741160Z" />
    <EventRecordID>10117</EventRecordID>
    <Correlation />
    <Execution ProcessID="13660" ThreadID="13664" />
    <Channel>Application</Channel>
    <Computer>DESKTOP-7DH039P</Computer>
    <Security UserID="S-1-5-21-331008269-550995778-4289027591-1001" />
  </System>
  <EventData>
    <Data Name="AppName">uad-ng-windows.exe</Data>
    <Data Name="AppVersion">0.0.0.0</Data>
    <Data Name="AppTimeStamp">660187c7</Data>
    <Data Name="ModuleName">nvoglv64.dll_unloaded</Data>
    <Data Name="ModuleVersion">31.0.15.5186</Data>
    <Data Name="ModuleTimeStamp">65f0dfad</Data>
    <Data Name="ExceptionCode">c0000005</Data>
    <Data Name="FaultingOffset">0000000000fbc5cc</Data>
    <Data Name="ProcessId">0x30e4</Data>
    <Data Name="ProcessCreationTime">0x1da82729c4cddd7</Data>
    <Data Name="AppPath">C:\Users\Алексей\Desktop\uad-ng-windows.exe</Data>
    <Data Name="ModulePath">nvoglv64.dll</Data>
    <Data Name="IntegratorReportId">7f014af8-f7e3-4e6b-b7ea-f6a928f3d6c4</Data>
    <Data Name="PackageFullName">
    </Data>
    <Data Name="PackageRelativeAppId">
    </Data>
  </EventData>
</Event>

AnonymousWP, 2024-03-30:
Those are Windows logs. I asked for UAD-ng logs, see https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/FAQ#where-are-the-logfiles.

Satorllous, 2024-03-31:
[UAD_20240331.log](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/files/14814109/UAD_20240331.log)

AnonymousWP, 2024-03-31:
It clearly says ADB isn't found in the logs, [so install ADB](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/Getting-started). However, since 1.0.3 you should get presented that ADB isn't installed instead of UAD-ng not starting. @edgy-b Any idea? See the video.

edgy-b, 2024-04-07:
> It clearly says ADB isn't found in the logs, [so install ADB](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/Getting-started). However, since 1.0.3 [...]

Hi. It tries to display, but fails to display on wgpu side, not sure what's going on. Not sure I'll be even able to reproduce, to check whats going on.

xe1st, 2024-04-13:
> ### Describe the bug
>
> I run the program, it closes immediately after opening, there are no errors https://youtu.be/k3JDSEtETO0
> ### Expected behavior
>
> fix
> ### You have a solution?
>
> No response
> ### [...]

Hi you missed the adb step I believe. Easiest way to do is:

1. download [android platform tools](https://dl.google.com/android/repository/platform-tools-latest-windows.zip) and unzip it (extract here)
2. Then download uad ng, extract it and put the uad-ng.exe inside platform tools folder and launch from there

AnonymousWP, 2024-04-13:
@Satorllous Please follow-up on this or else it'll be closed.

AnonymousWP, 2024-09-15:
Closing due to the issue being stale.

closed 2024-09-15 by AnonymousWP