Universal-Debloater-Alliance/universal-android-debloater-next-generation #398: pkg(scope):

oldman20, 2024-03-31:
### Describe the problem

Android Q
Samsung N960F

Force Close Galaxy App after few seconds opened if debloat *com.google.android.overlay.gmsconfig*

popup UI revert similiar OASP UI if debloat *com.google.android.overlay.modules.permissioncontroller*

### You have a solution?

Not select 
*com.google.android.overlay.gmsconfig*
*com.google.android.overlay.modules.permissioncontroller*

 to debloat

### Provide logs

[UAD_20240327.log](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/files/14814550/UAD_20240327.log)

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

aryanpednekar45, 2024-04-01:
I have them debloated on my S23 Ultra, doesn't seem to have this issue!
Old OneUI might be the issue!

oldman20, 2024-04-02:
so that maybe, cause im still OneUI 2.5

oldman20, 2024-04-05:
another issue on A10:
dont disable/ remove *framework-res__auto_generated_rro_vendor.apk* in `/vendor/overlay` folder, it will make headphone jack 3.5 not work

AnonymousWP, 2024-04-13:
> popup UI revert similiar OASP UI if debloat com.google.android.overlay.modules.permissioncontroller

This description isn't clear to me. Be more specific please.

> another issue on A10: dont disable/ remove _framework-res__auto_generated_rro_vendor.apk_ in `/vendor/overlay` folder, it will make headphone jack 3.5 not work

Make a new issue for this.

AnonymousWP, 2024-05-29:
@oldman20 Please follow-up on this.

oldman20, 2024-05-29:
> @oldman20 Please follow-up on this.

> This description isn't clear to me. Be more specific please.
>
>
> Make a new issue for this.

Sorry im busy, add this too, pls

oldman20, 2024-05-29:
> > popup UI revert similiar OASP UI if debloat com.google.android.overlay.modules.permissioncontroller
> 
> This description isn't clear to me. Be more specific please.

i mean when disble it, all popup ui will back to center screen appear instead bottom, and has The gray ui is very ugly like on aosp rom

bornasalman, 2024-05-30:
Yes, it's an overlay. It's in the package name. All overlays are reskins of the AOSP versions. It doesn't affect anything but may cause more battery drain as the AOSP scale and colour scheme is less efficient than the overlay versions.

AnonymousWP, 2024-05-30:
Thanks both of you, I'll fix that in the linked PR.

closed 2024-06-04 by AnonymousWP