Universal-Debloater-Alliance/universal-android-debloater-next-generation #417: Realme 6 pro softwere broke

youssefam1n, 2024-04-10:
### Describe the problem

after using the recommended settings my phone broke and i cant open it 

### You have a solution?

i had to hard reset the phone and it worked

### Provide logs

[UAD_20240127.log](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/files/14927117/UAD_20240127.log)
[UAD_20240410.log](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/files/14927119/UAD_20240410.log)
[uad_lists.json](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/files/14927120/uad_lists.json)

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Blitz517, 2024-04-10:
check if it not caused by com.qualcomm.qti.server.wigig.tethering.rro
or com.google.android.overlay...

schbrns, 2024-11-24:
@youssefam1n Could you please follow-up on this? Thanks.

closed 2025-06-05 by schbrns