Universal-Debloater-Alliance/universal-android-debloater-next-generation #423: pkg(scope): Pixel 6a not working properly after uninstalling recommended apps

xip0enix, 2024-04-13:
### Describe the problem

After I got rid of some recommended and advanced apps, like Chrome, in the past without any trouble, I thought I'd take it a step further and uninstall all the recommended ones, leaving just Maps, Messages, Drive, Photos, and eSIM on my Pixel 6a. But, after I did that, my phone stopped working. It wouldn't shut down or restart, and the Settings app kept crashing or showing a blank screen. Uninstalling apps also stopped working, and the uad debloater couldn't find my phone. Luckily, I stumbled upon the repair option for my Pixel, and that fixed everything. I'm not sure what caused the issue, but it might be worth checking out.

### Provide logs

[UAD_20240409.log](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/files/14967085/UAD_20240409.log)

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-04-13:
Do you know which specific debloat action caused this? Which app for example? @Blitz517 Perhaps you know.

Blitz517, 2024-04-13:
i can only guess which one did that
com.google.android.networkstack.tethering.overlay2021
com.google.android.connectivity.resources.overlay
com.google.android.overlay.permissioncontroller