Universal-Debloater-Alliance/universal-android-debloater-next-generation #432: Crash: Panick in main: called Option::unwrap() on a None value

xelxebar, 2024-04-27:
### Describe the bug

## Overview

The Nix package for 1.0.3 is crashing with the below error. Is anyone here able to reproduce?

```
[nix-shell:~]$ RUST_BACKTRACE=full ./uag-ng.sh
2024-04-27 17:57:14 WARN  [/build/universal-android-debloater-1.0.3-vendor.tar.gz/winit/src/platform_impl/linux/x11/xdisplay.rs:124] error setting XSETTINGS; Xft options won't reload automatically
2024-04-27 17:57:14 WARN  [/build/universal-android-debloater-1.0.3-vendor.tar.gz/wgpu-hal/src/gles/egl.rs:788] EGL_MESA_platform_surfaceless not available. Using default platform
thread 'main' panicked at /build/universal-android-debloater-1.0.3-vendor.tar.gz/wgpu-hal/src/gles/egl.rs:789:88:
called `Option::unwrap()` on a `None` value
stack backtrace:
   0:     0x5564f3e0dee8 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h83d3e31f1809523c
   1:     0x5564f3ac5ac0 - core::fmt::write::h76573a5842288faa
   2:     0x5564f3e23b14 - std::io::Write::write_fmt::hb095d534967ab243
   3:     0x5564f3e0e95e - std::sys_common::backtrace::print::h73765b495122298a
   4:     0x5564f3e0e655 - std::panicking::default_hook::{{closure}}::hfad133b7877f7d1a
   5:     0x5564f3e0f4ce - std::panicking::rust_panic_with_hook::h03308f3977d70e82
   6:     0x5564f3e0ef88 - std::panicking::begin_panic_handler::{{closure}}::h6543c8dd74f459e9
   7:     0x5564f3e0ef16 - std::sys_common::backtrace::__rust_end_short_backtrace::h68617e661e847078
   8:     0x5564f3e0ef03 - rust_begin_unwind
   9:     0x5564f3abca54 - core::panicking::panic_fmt::hf7f647d933c4266a
  10:     0x5564f3abda72 - core::panicking::panic::h423064a40da58821
  11:     0x5564f3ac85d5 - core::option::unwrap_failed::hffb1e9e838106d36
  12:     0x5564f4164a68 - wgpu_core::instance::Instance::new::h45a023e4468c6ba4
  13:     0x5564f3a489c5 - futures_executor::local_pool::block_on::h81a2843031214e94
  14:     0x5564f3f2b4ba - <iced_renderer::compositor::Compositor as iced_graphics::compositor::Compositor>::new::h11f9aa80284f287a
  15:     0x5564f3f97482 - iced_winit::application::run::hb1ecb9c33ec37eba
  16:     0x5564f3fd034e - uad_ng::main::h22da11ae92f257d4
  17:     0x5564f3f8eb87 - std::sys_common::backtrace::__rust_begin_short_backtrace::hd7031090c0c5e9e0
  18:     0x5564f3fcf593 - main
  19:     0x7f1835b7410e - __libc_start_call_main
  20:     0x7f1835b741c9 - __libc_start_main@@GLIBC_2.34
  21:     0x5564f3a47c65 - _start
  22:                0x0 - <unknown>
```

## Version Information

```
[nix-shell:~]$ nix-info --markdown --host-os
 - system: `"x86_64-linux"`
 - host os: `Linux 6.8.7, Guix System, noversion, nobuild`
 - multi-user?: `no`
 - sandbox: `yes`
 - version: `nix-env (Nix) 2.16.1`
 - channels(x): `"nixpkgs-23.11pre505289.57695599bdc"`
 - nixpkgs: `/home/x/.nix-defexpr/channels/nixpkgs`

[nix-shell:~]$ nix-env -qaP universal-android-debloater
nixpkgs.universal-android-debloater  universal-android-debloater-1.0.3
```

### Expected behavior

Application should boot.

### Provide logs

[UAD_20240427.log](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/files/15137699/UAD_20240427.log)

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

xelxebar, 2024-05-01:
@lavafroth Uh, I forgot to sanitize `./uad-ng.sh` out. It's a temporary wrapper fix.  Something at startup wants `libxkbcommon-x11.so` but the nix wrapper doesn't have it in `LD_LIBRARY_PATH`, causing initialization to fall over. I didn't intend to mix issues here, though.

Seeing the same panic on the `nix-flake` branch, unfortunately:

```
x@myyrix ~/universal-android-debloater-next-generation$ git rev-parse HEAD
1adf69bd8d358c6393bd487ef56bee9bb0924ff2
x@myyrix ~/universal-android-debloater-next-generation$ cargo run
    Finished dev [unoptimized + debuginfo] target(s) in 0.24s
     Running `target/debug/uad-ng`
2024-05-01 16:58:29 WARN  [/home/x/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/x11/xdisplay.rs:124] error setting XSETTINGS; Xft options won't reload automatically
xkbcommon: ERROR: /home/x/.XCompose:83:25: unrecognized keysym "circumflex" on left-hand side
2024-05-01 16:58:29 WARN  [/home/x/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/egl.rs:788] EGL_MESA_platform_surfaceless not available. Using default platform
thread 'main' panicked at /home/x/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/egl.rs:789:88:
called `Option::unwrap()` on a `None` value
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Frigyes06, 2024-05-03:
If I read that log correctly, this panic happens inside the wgpu-hal crate however. So maybe it'd be beneficial to open an issue for them?

xelxebar, 2024-05-06:
@Frigyes06 Maybe? It's happening at the line below in `wgpu-hal`. They're explicitly performing an `unsafe` operation which I assume is intentional. Possible that we're not correctly setting up the display environment or something?

```
            if let (Some(library), Some(egl)) = (wayland_library, egl1_5) {
            ...
            } else {
                log::warn!("EGL_MESA_platform_surfaceless not available. Using default platform");
                let display = unsafe { egl.get_display(khronos_egl::DEFAULT_DISPLAY) }.unwrap();   // <-- this line
                (display, None, WindowKind::Unknown)
            };
```

AnonymousWP, 2024-09-15:
@xelxebar Is this issue still relevant? Because of https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/pull/421.

Frigyes06, 2024-11-26:
Is this still an issue @xelxebar ?

AnonymousWP, 2025-01-13:
@xelxebar Sorry that it took this long. Could you try with one of these artifacts? https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/actions/runs/12749084042

medmedin2014, 2025-04-01:
The crash also happens with version 1.1.2 on:
```
Operating System: Manjaro Linux 
KDE Plasma Version: 6.3.3
KDE Frameworks Version: 6.12.0
Qt Version: 6.8.2
Kernel Version: 6.12.20-2-MANJARO (64-bit)
Graphics Platform: Wayland
```

It shows the following errors:
```
./uad-ng-linux 
2025-04-01 19:53:10 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/vulkan/instance.rs:291] Unable to find extension: VK_KHR_surface
2025-04-01 19:53:10 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/vulkan/instance.rs:291] Unable to find extension: VK_KHR_xlib_surface
2025-04-01 19:53:10 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/vulkan/instance.rs:291] Unable to find extension: VK_KHR_xcb_surface
2025-04-01 19:53:10 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/vulkan/instance.rs:291] Unable to find extension: VK_KHR_wayland_surface
2025-04-01 19:53:10 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/vulkan/instance.rs:291] Unable to find extension: VK_EXT_swapchain_colorspace
2025-04-01 19:53:10 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/vulkan/instance.rs:291] Unable to find extension: VK_KHR_get_physical_device_properties2
thread 'main' panicked at /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-core-0.19.4/src/instance.rs:521:39:
called `Option::unwrap()` on a `None` value
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
zsh: IOT instruction (core dumped)  ./uad-ng-linux
```

Rudxain, 2025-04-03:
Already fixed here: gfx-rs/wgpu#5166. We have to upgrade Iced to get the patch for this app

AnonymousWP, 2025-04-28:
> Already fixed here: [gfx-rs/wgpu#5166](https://github.com/gfx-rs/wgpu/pull/5166). We have to upgrade Iced to get the patch for this app

Can you open a new PR where you bump all dependencies? Should be a single command. See https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/623

Rudxain, 2025-04-28:
> Can you open a new PR where you bump all dependencies? Should be a single command. See [#623](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/623)

The problem is that the `wgpu` patch is available since Iced 0.13, so `cargo update` is not enough. Bumping (other deps) would still be a good idea anyways

AnonymousWP, 2025-04-28:
> The problem is that the `wgpu` patch is available since Iced 0.13, so `cargo update` is not enough. Bumping (other deps) would still be a good idea anyways

Yes as we agreed upon bumping all deps starting with a new release. Could you please do that?

Rudxain, 2025-04-29:
Oh ok, I'll prepare a draft PR

medmedin2014, 2025-04-30:
@Rudxain  @AnonymousWP 
Any workaround to use the app? it's been a month and can't seem to launch it.

Rudxain, 2025-05-01:
This environment-variable might help:
```sh
export WGPU_BACKEND=gl
```

However, updating the `flake.lock` might also help

medmedin2014, 2025-05-01:
> This environment-variable might help:
> 
> export WGPU_BACKEND=gl

I tested setting `WGPU_BACKEND=gl` both on Manjaro and Arch, and they returned the same error:
```
env WGPU_BACKEND=gl ./uad-ng-linux
thread 'main' panicked at /home/runner/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/wgpu-core-0.19.4/src/instance.rs:521:39:
called `Option::unwrap()` on a `None` value
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
zsh: IOT instruction (core dumped)  env WGPU_BACKEND=gl ./uad-ng-linux
```
> 
> Updating the `flake.lock` might also help

I don't how to update this feature, plus Arch based distros are continuously updating their packages to latest ones.

Rudxain, 2025-05-02:
I've just read that you're using Manjaro 😅. The flakes only apply if you use the Nix package-manager

medmedin2014, 2025-05-20:
> I've just read that you're using Manjaro 😅. The flakes only apply if you use the Nix package-manager

So what is the correct way to rebuild or launch correctly this app on Arch based distros?

Rudxain, 2025-05-20:
Sadly, not much can be done. A short-term solution would be to soft-fork Iced at its v12 tag, then backport the wgpu patch.

I'd rather rewrite the UADNG front-end code than to backport stuff, because that code is a mess anyways, so wasting time backporting might not be worth it.

Another solution is #746, which seems easier.

These days, I don't have time for implementing either :(

medmedin2014, 2025-11-15:
So more than a year and half and no fix for this critical error ?

AnonymousWP, 2025-11-16:
Very likely related to https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/pull/1064, please read through there.