Universal-Debloater-Alliance/universal-android-debloater-next-generation #438: bug(scope): Startup on crash

EnderNon, 2024-04-30:
### Describe the bug

Ran the program in linux, I get an error in terminal.

### Expected behavior

the app to run

### Provide logs

[UAD_20240430.log](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/files/15170062/UAD_20240430.log)

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

EnderNon, 2024-04-30:
oh right also, I'm running ArchLinux with KDE Plasma, on Linux Zen kernel

edgy-b, 2024-05-01:
> > oh right also, I'm running ArchLinux with KDE Plasma, on Linux Zen kernel
> 
> @EnderNon Duplicate #383

closed 2024-05-03 by AnonymousWP