Universal-Debloater-Alliance/universal-android-debloater-next-generation #441: After using the tool, and going through things carefully, everything works except; my phone no longer connects to 4G under any circumstances

ZohZie, 2024-05-01:
### Describe the problem

Pixel 7 pro.

As the title says, after restarting everything works fine and feels faster/battery longer.

Overall extremely happy, as I don't leave the house much I rarely experienced mobile data.

But I have now noticed that in 4G and 5G strong areas, where my partners pixel 8 pro gets 5 and 4g fully, mine only gets LTE.

[image]

These are my settings, I have tried to reset the network settings on my phone. Rebooted ect.

It is on a permanent LTE of 2 or 3 bars, when it should be full bar 4G.

### Provide logs

Unfortunately lost due to a pc reset, due to a boot file corruption.

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

ZohZie, 2024-05-01:
[image]

ZohZie, 2024-05-01:
i redownloaded the app, and these are listed under "disabled"

com.android.bips.auto_generated_rro_product__
com.android.bluetoothmidiservice
com.android.calllogbackup
com.android.captiveportallogin.overlay
com.android.cellbroadcastreceiver.overlay.pixel
com.android.cellbroadcastservice.overlay.pixel
com.android.egg
com.android.hotwordenrollment.xgoogle
com.android.hotwordenrollment.okgoogle
com.android.managedprovisioning.overlay
com.android.nfc.auto_generated_rro_product__
com.android.nfc.auto_generated_rro_vendor__
com.android.omadm.service.auto_generated_rro_product__
com.android.omadm.service.auto_generated_rro_vendor__
com.android.remoteprovisioner
com.android.safetyregulatoryinfo.auto_generated_rro_product__
com.android.sdm.plugins.diagmon
com.android.simappdialog.auto_generated_rro_product__
com.android.systemui.plugin.globalactions.wallet
com.android.traceur.auto_generated_rro_product__
com.android.traceur.auto_generated_rro_vendor__
com.google.ambient.streaming
com.google.android.accessibility.soundamplifier
com.google.android.adservices.api
com.google.android.apps.accessibility.voiceaccess
com.google.android.apps.carrier.log
com.google.android.apps.cbrsnetworkmonitor
com.google.android.apps.healthdata
com.google.android.apps.helprtc
com.google.android.apps.internal.betterbug

com.google.android.apps.recorder
com.google.android.apps.retaildemo.preload
com.google.android.apps.safetyhub
com.google.android.apps.scone
com.google.android.apps.security.securityhub
com.google.android.apps.turbo
com.google.android.apps.wearables.maestro.companion
com.google.android.carrier
com.google.android.carriersetup
com.google.android.documentsui.theme.pixel
com.google.android.euiccoverlay
com.google.android.factoryota
com.google.android.federatedcompute
com.google.android.feedback
com.google.android.gms.location.history
com.google.android.grilservice
com.google.android.odad
com.google.android.ondevicepersonalization.services
com.google.android.pixel.setupwizard.auto_generated_rro_product__
com.google.android.pixel.setupwizard.overlay
com.google.android.pixel.setupwizard.overlay2019
com.google.android.pixel.setupwizard.overlay2021
com.google.android.projection.gearhead
com.google.android.soundpicker
com.google.android.tag
com.google.android.turboadapter
com.google.android.videos
com.google.ar.core
com.google.audio.hearing.visualization.accessibility.scribe
com.google.mainline.adservices
com.google.pixel.digitalkey.timesync
com.samsung.slsi.telephony.oem.oemrilhookservice
com.vzw.apnlib

ZohZie, 2024-05-01:
i re-installed all of them, and it immediately got 4G back.

now i need help to figure out which one caused this.

also, because i reset network settings, how owuld one get those back? they are backed up as on an hour ago

ZohZie, 2024-05-01:
it is seemingly com.google.android.carrier or com.google.android.apps.scone

ZohZie, 2024-05-01:
it is confirmed as com.google.android.carrier

i seemingly lost the backup, unrelated to the issue. Closing as it is solved.

closed 2024-05-01 by ZohZie

thyTwilightGoth, 2024-05-14:
Not reopening the issue, (I hope) but I can also confirm that disabling the above mentioned package on my Pixel 7a breaks calling and texting unless I disable the internet.