Universal-Debloater-Alliance/universal-android-debloater-next-generation #444: bug(software startup): UAD's UI doesn't show up on my laptop although its service is still running

itsvngo, 2024-05-02:
### Describe the bug

I installed UAD on Windows 10 LTSC build. The tool had worked a week ago on the same build. But i had deleted it along with adb for personal reason. 
Today, i redownloaded the tool and adb, I gave path to adb, driver for my pixel 5 was already installed, check for adb devices, adb was working (the proof was pixel flasher working fine).
I moved UAD tool to adb folder, started it up, nothing happened, checked the task manager, the tool's service was already working, but the UI didnt come up. 

### Expected behavior

The UI comes up as usual so i can interact with  the tool

### You have a solution?

I tried to make a fresh adb folder to start the tool but nothing happens, i even started it in administration mode. I tried the OG version of 0x192 too but still nothing. I don't know how it had just worked a week ago but now nothing shows up

### Provide logs

[UAD_20240427.log](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/files/15191103/UAD_20240427.log)
[UAD_20240502.log](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/files/15191105/UAD_20240502.log)
[uad_lists.json](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/files/15191106/uad_lists.json)

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

xe1st, 2024-05-05:
Same for me on Win10 LTSC. The app worked when I tried it about a month ago or so. It also worked on Linux. So I think this is an LTSC specific issue. Adb.exe runs in task manager but GUI does not show up

log
```
2024-05-05 13:54:17 ERROR [src\core\config.rs:79] Failed to read config file: `The system cannot find the file specified. (os error 2)`
2024-05-05 13:54:17 ERROR [src\core\config.rs:81] Restoring default config file
```

NenoIT, 2024-05-05:
Same for me on Windows 11, I can see the process in Task Manager but the GUI dosen't show up. I tried all sort of thing like the OP said but nothing, unfortunately I don't have logs since I deleted all the files.

edgy-b, 2024-05-05:
> Same for me on Win10 LTSC. The app worked when I tried it about a month ago or so. It also worked on Linux. So I think this is an LTSC specific issue, maybe it is due to recent updates or anything Adb.exe runs in task manager but GUI does not show up. Ping me if you guys can find anything.
> 
> log
> 
> ```
> 2024-05-05 13:54:17 ERROR [src\core\config.rs:79] Failed to read config file: `The system cannot find the file specified. (os error 2)`
> 2024-05-05 13:54:17 ERROR [src\core\config.rs:81] Restoring default config file
> ```

It should load new config file, but just in case try to rename config file in `C:\Users\<user>\AppData\Roaming\uad` and if it loads new file

xe1st, 2024-05-05:
nevermind it started working for some reason lol.

edgy-b, 2024-05-05:
@itsvngo try connect phone to pc without programm running. Open shell and type `adb devices`, any devices listed?

zmileski, 2024-05-14:
I had same problem.
just opened with integrated graphic instead of nvidia and its worked.
Dont know why but now its working 
dont know if u have nvidia or not 
but  something is with graphics maybe driver i dont know.

AnonymousWP, 2024-09-15:
Closing this due to the issue being stale.

closed 2024-09-15 by AnonymousWP