Universal-Debloater-Alliance/universal-android-debloater-next-generation #448: Cant remove system apps on iqoo phone

hydrophonic-droid, 2024-05-04:
### Describe the bug

uad cannot remove the system apps on my iqoo z9. I searched for the app and pressed uninstall but nothing happened.the app I was trying to remove was v appstore.

### Expected behavior

file should have been uninstalled or disabled but neither happened. 

### Provide logs

[UAD_20240503.log](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/files/15209545/UAD_20240503.log)
[UAD_20240504.log](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/files/15209546/UAD_20240504.log)
[uad_lists.json](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/files/15209547/uad_lists.json)

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

edgy-b, 2024-05-06:
This is not programm bug, but rather vivo restricting uninstallation of some packages.
By reading this post it's possible to uninstall manually:
https://xdaforums.com/t/bypass-delete_failed_user_restricted-error-funtouch-os-13.4570979/
And this russian site explains that vivo decided to add `vivo.uninstall.compatibility.not_adbUninstall` to apps.
Translate in your browser to preffered language https://dtf.ru/u/523140-danila-romanov/1154461-eshe-odin-sposob-udaleniya-predustanovlennyh-prilozhenii-android-12

boyedarat, 2024-05-12:
> This is not programm bug, but rather vivo restricting uninstallation of some packages. By reading this post it's possible to uninstall manually: [...]

Unfortunately these do not work with Funtouch OS 14.

AnonymousWP, 2024-09-15:
Closing as @edgy-b already gave a solution/explanation.

closed 2024-09-15 by AnonymousWP