Universal-Debloater-Alliance/universal-android-debloater-next-generation #451: pkg(scope): com.oplus.nas (Network Services)

aryanpednekar45, 2024-05-06:
### List your packages

- Package name: Network Services (com.oplus.nas)
- List: OEM
- Removal: Recommended/Advanced
- Description: Has Location & Phone permission, can read your Call logs. Runs in background
After Disabling, no issues occured (OxygenOS 14, OnePlus 11)

### Acknowledgements

- [X] I removed all those packages on my phone. If not, why? Explain in the issue.
- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Blitz517, 2024-05-06:
about de-bloater f-droid I mention it, because its easy to recover back apps using twrp

AnonymousWP, 2024-05-21:
Why would it be `Advanced`? What is its purpose? 👀

closed 2024-05-21 by AnonymousWP