Universal-Debloater-Alliance/universal-android-debloater-next-generation #453: i mistakenly disabled or uninstalled app on universal android debloater.

upfringe, 2024-05-07:
### Describe the problem

now most of the apps are gone. wifi is working but the sim is not showing. i am trying to enable the apps using the debloater but it enableing but nothing happening . on refresh it goes back to disabled.
please help me. i am unable to restore it. i cannot type , i cannot access files. but i can still use facebook, messenger and some other app. also browser is uninstalled.

android device: Redmi4

i have partial log below

### Provide logs

2024-05-07 22:40:18 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.facebook.system -> Package com.facebook.system new state: enabled
2024-05-07 22:40:18 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.google.android.apps.maps -> Package com.google.android.apps.maps new state: enabled
2024-05-07 22:40:18 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.google.android.apps.docs -> Package com.google.android.apps.docs new state: enabled
2024-05-07 22:40:18 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.google.android.apps.maps -> Package com.google.android.apps.maps new state: enabled
2024-05-07 22:40:18 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.google.android.apps.photos -> Package com.google.android.apps.photos new state: enabled
2024-05-07 22:40:19 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.google.android.feedback -> Package com.google.android.feedback new state: enabled
2024-05-07 22:40:19 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.google.android.apps.photos -> Package com.google.android.apps.photos new state: enabled
2024-05-07 22:40:19 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.google.android.apps.tachyon -> Package com.google.android.apps.tachyon new state: enabled
2024-05-07 22:40:19 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.google.android.apps.tachyon -> Package com.google.android.apps.tachyon new state: enabled
2024-05-07 22:40:19 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.google.android.gm -> Package com.google.android.gm new state: enabled
2024-05-07 22:40:19 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.google.android.gm -> Package com.google.android.gm new state: enabled
2024-05-07 22:40:19 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.google.android.feedback -> Package com.google.android.feedback new state: enabled
2024-05-07 22:40:19 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.google.android.marvin.talkback -> Package com.google.android.marvin.talkback new state: enabled
2024-05-07 22:40:19 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.google.android.music -> Package com.google.android.music new state: enabled
2024-05-07 22:40:19 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.google.android.marvin.talkback -> Package com.google.android.marvin.talkback new state: enabled
2024-05-07 22:40:19 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.google.android.music -> Package com.google.android.music new state: enabled
2024-05-07 22:40:20 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.google.android.talk -> Package com.google.android.talk new state: enabled
2024-05-07 22:40:20 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.google.android.onetimeinitializer -> Package com.google.android.onetimeinitializer new state: enabled
2024-05-07 22:40:20 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.google.android.talk -> Package com.google.android.talk new state: enabled
2024-05-07 22:40:20 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.google.android.onetimeinitializer -> Package com.google.android.onetimeinitializer new state: enabled
2024-05-07 22:40:20 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.google.android.videos -> Package com.google.android.videos new state: enabled
2024-05-07 22:40:20 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.huaqin.diaglogger -> Package com.huaqin.diaglogger new state: enabled
2024-05-07 22:40:20 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.google.android.videos -> Package com.google.android.videos new state: enabled
2024-05-07 22:40:20 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.google.android.youtube -> Package com.google.android.youtube new state: enabled
2024-05-07 22:40:20 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.google.android.youtube -> Package com.google.android.youtube new state: enabled
2024-05-07 22:40:20 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.huaqin.factory -> Package com.huaqin.factory new state: enabled
2024-05-07 22:40:20 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.huaqin.diaglogger -> Package com.huaqin.diaglogger new state: enabled
2024-05-07 22:40:21 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.mi.AutoTest -> Package com.mi.AutoTest new state: enabled
2024-05-07 22:40:21 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.mi.android.globalpersonalassistant -> Package com.mi.android.globalpersonalassistant new state: enabled
2024-05-07 22:40:21 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.huaqin.factory -> Package com.huaqin.factory new state: enabled
2024-05-07 22:40:21 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.mi.AutoTest -> Package com.mi.AutoTest new state: enabled
2024-05-07 22:40:21 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.mi.android.globalpersonalassistant -> Package com.mi.android.globalpersonalassistant new state: enabled
2024-05-07 22:40:21 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.mi.webkit.core -> Package com.mi.webkit.core new state: enabled
2024-05-07 22:40:21 ERROR [src/core/sync.rs:129] [Recommended] pm enable --user 0 com.mi.globalTrendNews -> Error: java.lang.SecurityException: Shell cannot change component state for com.mi.globalTrendNews/null to 1
2024-05-07 22:40:21 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.mi.globalTrendNews -> Package com.mi.globalTrendNews new state: enabled
2024-05-07 22:40:21 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.mipay.wallet.id -> Package com.mipay.wallet.id new state: enabled
2024-05-07 22:40:21 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.mipay.wallet.id -> Package com.mipay.wallet.id new state: enabled
2024-05-07 22:40:21 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.mi.webkit.core -> Package com.mi.webkit.core new state: enabled
2024-05-07 22:40:22 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.mipay.wallet.in -> Package com.mipay.wallet.in new state: enabled
2024-05-07 22:40:22 ERROR [src/core/sync.rs:129] [Recommended] pm enable --user 0 com.mipay.wallet.in -> Error: java.lang.SecurityException: Shell cannot change component state for com.mipay.wallet.in/null to 1
2024-05-07 22:40:22 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.miui.antispam -> Package com.miui.antispam new state: enabled
2024-05-07 22:40:22 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.miui.analytics -> Package com.miui.analytics new state: enabled
2024-05-07 22:40:22 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.miui.backup -> Package com.miui.backup new state: enabled
2024-05-07 22:40:22 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 0 com.miui.antispam -> Package com.miui.antispam new state: enabled
2024-05-07 22:40:22 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.miui.backup -> Package com.miui.backup new state: enabled
2024-05-07 22:40:22 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.miui.calculator -> Package com.miui.calculator new state: enabled
2024-05-07 22:40:22 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.miui.bugreport -> Package com.miui.bugreport new state: enabled
2024-05-07 22:40:22 INFO  [src/core/sync.rs:123] [Recommended] pm enable --user 999 com.miui.analytics -> Package com.miui.analytics new state: enabled
[... cut ...]
2024-05-07 22:40:49 INFO  [src/gui/views/list.rs:144] -----------------------------------------------------------------
2024-05-07 22:40:49 INFO  [src/gui/views/list.rs:145] ANDROID_SDK: 24 | DEVICE: xiaomi Redmi Note 4
2024-05-07 22:40:49 INFO  [src/gui/views/list.rs:149] -----------------------------------------------------------------

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Blitz517, 2024-05-07:
Only disabling apps work? rebooted phone?

Blitz517, 2024-05-07:
my guesses its caused by com.qualcomm.qti.loadcarrier

upfringe, 2024-05-10:
Nothing worked, so I factory reset the phone. Now all the data is gone. I couldn't even access the file manager. Earlier, I used to do it with [0x192/universal-android-debloater](https://github.com/0x192/universal-android-debloater) and it seemed fine, but now I used this next-gen one, and boom, everything disappeared. I don't know how I managed to do that, lol. Anyway thanks for the reply

closed 2024-05-10 by upfringe

Blitz517, 2024-05-10:
i wanna know which app caused it if its uad-ng issue