Universal-Debloater-Alliance/universal-android-debloater-next-generation #458: bug(scope): note on documentsui and why you should not delete it (google files is different)

equwal, 2024-05-10:
### Describe the bug

Just wanted to quickly say what documentsui is and why you shouldn't turn it off.

Documentsui is the standard API access to a file picker, and is essentially a simplified version of google files. Many apps will cease to work without it, such as the "Voice" audiobook player or the Tachiyomi manga viewer. Any app without its own file picker, that attempts to use documentsui instead of calling "choose file manager", will not work.

You can still delete google files, which is a separate app.

### Expected behavior

You want to delete google files bl0at and install something like Simple File Manager, so you mistakenly uninstall documentsUI.

### You have a solution?

Do not turn off documentsui

### Provide logs

no

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-05-21:
You used the wrong template. It's not a bug at all.

AnonymousWP, 2024-05-27:
That's already documented properly: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/main/resources/assets/uad_lists.json#L7712. 🙂

closed 2024-05-27 by AnonymousWP