Universal-Debloater-Alliance/universal-android-debloater-next-generation #459: bug(scope): EGL says it can present to the window but not natively

MLSci, 2024-05-13:
### Describe the bug

Under Wayland:
```
01:19:25 WARN [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/egl.rs:268] No config found! 
01:19:25 WARN [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/egl.rs:257] EGL says it can present to the window but not natively
```
The UAD windows is pure black and shows nothing.

Under X11:
Besides the above two warnings -
```
2024-05-13 01:26:54 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/mod.rs:713] X11 error: XError { description: "BadMatch (invalid parameter attributes)", error_code: 8, request_code: 149, minor_code: 4, } thread 'main' panicked at /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/x11/window.rs:1208:18: Failed to call XMapRaised: XError { description: "BadMatch (invalid parameter attributes)", error_code: 8, request_code: 149, minor_code: 4 } note: run with RUST_BACKTRACE=1 environment variable to display a backtrace 
thread 'main' panicked at /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/x11/window.rs:601:34: called Result::unwrap() on an Err value: PoisonError { .. } 
stack backtrace: 
0: 0x55ccf766a6a8 - 
1: 0x55ccf72ff970 - 
2: 0x55ccf763493e - 
3: 0x55ccf766c17e - 
4: 0x55ccf766baa5 - 
5: 0x55ccf766c9b7 - 
6: 0x55ccf766c4e0 - 
7: 0x55ccf766c436 - 
8: 0x55ccf766c423 - 
9: 0x55ccf72fdac4 - 
10: 0x55ccf7305132 - 
11: 0x55ccf7a440e9 - 
12: 0x55ccf7a4ddec - 
13: 0x55ccf7773384 - 
14: 0x55ccf77e2a7c - 
15: 0x55ccf77d9fc3 - 
16: 0x55ccf781084f - 
17: 0x55ccf77cecfc - 
18: 0x55ccf780f9a4 - 
19: 0x7fc8482cb14a - __libc_start_call_main 
20: 0x7fc8482cb20b - __libc_start_main_alias_2 
21: 0x55ccf728bc65 - 
22: 0x0 - thread 'main' panicked at library/core/src/panicking.rs:163:5: panic in a destructor during cleanup thread caused non-unwinding panic. aborting. 
fish: Job 1, './uad-ng-linux' terminated by signal SIGABRT (Abort)
```
System Info:
Fedora 39 in distrobox
KDE Plasma Version: 6.0.4
KDE Frameworks Version: 6.1.0
Qt Version: 6.7.0
Kernel Version: 6.8.9-300.fc40.x86_64 (64-bit)
Processors: 4 × Intel® Core™ i5-7500 CPU @ 3.40GHz
Graphics Processor: NVIDIA GeForce GTX 1050 Ti/PCIe/SSE2 (v550 driver)

### Expected behavior

The UAD program windows shows normally, either in Wayland or X11.

### Provide logs

```
2024-05-13 01:09:01 ERROR [src/core/config.rs:79] Failed to read config file: `No such file or directory (os error 2)`
2024-05-13 01:09:01 ERROR [src/core/config.rs:81] Restoring default config file
2024-05-13 01:09:01 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/egl.rs:268] No config found!
2024-05-13 01:09:01 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/egl.rs:257] EGL says it can present to the window but not natively
2024-05-13 01:09:01 DEBUG [src/core/update.rs:180] Checking for UAD-ng update
2024-05-13 01:09:01 INFO  [src/gui/views/list.rs:144] -----------------------------------------------------------------
2024-05-13 01:09:01 INFO  [src/gui/views/list.rs:145] ANDROID_SDK: 34 | DEVICE: samsung SM-X700
2024-05-13 01:09:01 INFO  [src/gui/views/list.rs:149] -----------------------------------------------------------------
2024-05-13 01:26:32 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/egl.rs:268] No config found!
2024-05-13 01:26:32 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/egl.rs:257] EGL says it can present to the window but not natively
2024-05-13 01:26:32 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/mod.rs:713] X11 error: XError {
    description: "BadMatch (invalid parameter attributes)",
    error_code: 8,
    request_code: 149,
    minor_code: 4,
}
2024-05-13 01:26:32 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/mod.rs:713] X11 error: XError {
    description: "BadMatch (invalid parameter attributes)",
    error_code: 8,
    request_code: 149,
    minor_code: 4,
}
2024-05-13 01:26:32 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/mod.rs:713] X11 error: XError {
    description: "BadMatch (invalid parameter attributes)",
    error_code: 8,
    request_code: 149,
    minor_code: 4,
}
2024-05-13 01:26:54 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/egl.rs:268] No config found!
2024-05-13 01:26:54 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.3/src/gles/egl.rs:257] EGL says it can present to the window but not natively
2024-05-13 01:26:54 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/mod.rs:713] X11 error: XError {
    description: "BadMatch (invalid parameter attributes)",
    error_code: 8,
    request_code: 149,
    minor_code: 4,
}
2024-05-13 01:26:54 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/mod.rs:713] X11 error: XError {
    description: "BadMatch (invalid parameter attributes)",
    error_code: 8,
    request_code: 149,
    minor_code: 4,
}
2024-05-13 01:26:54 ERROR [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/mod.rs:713] X11 error: XError {
    description: "BadMatch (invalid parameter attributes)",
    error_code: 8,
    request_code: 149,
    minor_code: 4,
}

```

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2025-01-09:
Might get fixed if we do #645

MLSci, 2025-01-15:
> @MLSci @big-chungus69420 Sorry that it took this long. Could you try with one of these artifacts? https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/actions/runs/12749084042

Thanks for the update. Unfortunately I've switch from Linux to Windows. All I can confirm is that UAD works flawlessly in Windows

Rudxain, 2025-01-16:
> UAD works flawlessly in Windows

Same hardware? If it is, then it's likely that the NVIDIA drivers work better on Windows

MLSci, 2025-02-20:
> Same hardware? 
Yep
> likely the NVIDIA drivers work better on Windows
I thought this is well known(?)

zukigay, 2025-08-16:
ran into this same issue, on newest build