Universal-Debloater-Alliance/universal-android-debloater-next-generation #463: When you first start UAD, Windows Defender detects a threat

Marekk2k, 2024-05-14:
### Describe the bug

When you start UAD for the first time, Windows Defender considers UAD to be a dangerous program and displays a window about protecting your computer against this tool. This is most likely due to the fact that the program, as Windows Defender shows, does not have a signed author when it shows "unknown publisher" under the publisher category.

My Windows version: 10 22H2

### Expected behavior

Potential threat message when UAD is first launched 

### You have a solution?

Author's signature so that Windows Defender does not recognize him as unknown 

### Provide logs

I don't think it's necessary...

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-05-15:
I think this should rather be a feature request instead of a bug report. The feature request should then be that we would like to sign the application.

Frigyes06, 2024-05-21:
https://stackoverflow.com/questions/48946680/how-to-avoid-the-windows-defender-smartscreen-prevented-an-unrecognized-app-fro/66582477#66582477

https://www.microsoft.com/en-us/wdsi/filesubmission

Frigyes06, 2024-05-21:
I can make a submission of incorrectly detected malware/PUA of you tell me:
- the security product you use
- detection name
- definition version
- a screenshot(s) of what windows says exactly

Marekk2k, 2024-05-22:
The message looks like this. This is what I had the first time when I ran UAD 1.0.3 after formatting Windows 10. I downloaded the photos from the Internet because this message no longer appears after clicking "Run anyway". but take my word for it. Windows Defender shows that the UAD file does not have a signed author, and this may be the cause of such a message 
[image]
[image]