Universal-Debloater-Alliance/universal-android-debloater-next-generation #468: pkg(scope): me.phh.treble.app

Byteintosh, 2024-05-20:
### List your packages

- Package name: PhhTrebleApp `me.phh.treble.app`
- List: Misc
- Removal: Unsafe
- Description: Used for settings GSI rom that based by Phh Treble. This should not be remove.

### Acknowledgements

- [X] I removed all those packages on my phone. If not, why? Explain in the issue.
- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Blitz517, 2024-05-20:
we want really custom rom to be listed?

AnonymousWP, 2024-05-21:
Hmm, we could. We could assign it to the `Oem` list. I believe there are already some packages there.

PR #481 (2024-05-27): pkg(me.phh.treble.app): add package for GSI ROM
- [pkg(me.phh.treble.app): add package for GSI ROM](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/commit/589ecbb9153187325413082fabde57d61c9b5a28)
- [pkg(com.android.settings.os.overlay): remove duplicate](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/commit/ad0d480fac3722486f0d6618aaec80771e02b9f9)

Fixes #468.
closed 2024-05-27 by AnonymousWP