Universal-Debloater-Alliance/universal-android-debloater-next-generation #479: pkg(scope): com.qualcomm.timeservice

bornasalman, 2024-05-26:
### List your packages

- Package name: com.qualcomm.timeservice
- Current list: Misc
  - Proposed list: Misc
- Current removal: Expert
  - Proposed removal: Expert
- Current description: 

"Qualcomm Time Service\nOccasionally runs in the background.\nCould be what syncs the CPU clock with Android time?\nProbably not something you want to disable."

  - Proposed description: 

"Qualcomm Time Service\nUpdates time-services user time offset when user changes time of the day and Android sends a TIME_CHANGED or DATE_CHANGED intents.\ntime-services restores the time of the day after reboot using this offset.\nhttps://github.com/bcyj/android_tools_leeco_msm8996/blob/master/time-services/src/com/qualcomm/timeservice/TimeServiceBroadcastReceiver.java\n"

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

bornasalman, 2024-05-26:
@AnonymousWP Wasn't sure if I should create a new issue. Edited title in case.

Package name: com.google.android.configupdater

Current removal: Advanced
Proposed removal: Expert

Current description:
"ConfigUpdater\nOccasionally runs in the background.\nAuto updates certificates for TLS connection, firewall configuration, e.t.c.\nMainly used for Google services? Might be fine to disable if you don't use Google services. Disabling might mess with security if you do use them though.\nhttps://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/os/ConfigUpdate.java"

Proposed description:
"Config Update\nIntents used to provide unbundled updates of system data.\nAll require the UPDATE_CONFIG permission.\n Updates: (1) system wide certificate pins for TLS connections, (2) system wide Intent firewall, (3) list of permium SMS short codes, (4) list of carrier provisioning URLs, (5) set of trusted logs used for Certificate Transparency support for TLS connections, (6) language detection model file, (7) smart selection model file, (8) conversation actions model file, (9) network watchlist config file, (10) intent action indicating that the updated carrier id config is available, (11) the emergency number database into the devices, (12) an integer to indicate the numeric version of the new data -- devices should only install if the update version is newer than the current one, (13) Hash of the database, which is encoded by base-16 SHA512.\nhttps://android.googlesource.com/platform/frameworks/base/+/refs/heads/master/core/java/android/os/ConfigUpdate.java\n"

AnonymousWP, 2024-05-28:
Thanks for this very detailed issue and documentation! I opened a PR. 🙂

closed 2024-05-30 by AnonymousWP