Universal-Debloater-Alliance/universal-android-debloater-next-generation #486: pkg(scope): Disabling com.hoffnung breaks lock screen notifications

ThatGreenBlur, 2024-05-30:
### List your packages

- Package name: com.hoffnung
- Current removal: Recommended
  - Proposed removal: Expert
- Current description: Remote Config Test. WARNING: you should DISABLE this package instead of uninstalling, because uninstalling causes screen flicker
  - Proposed description: Remote Config Test. WARNING: you should DISABLE this package instead of uninstalling, because uninstalling causes screen flicker, Disabling this also removes the ability to see notifications in your lock screen!

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Blitz517, 2024-05-30:
I think it may be in advanced, but only if uad-ng will block uninstall for some apps and only disable will be available

bornasalman, 2024-05-30:
Is this on the Infinix Note 10? These people were able to get a decompiled version of the APK over company email after some resistance. There's more going on with how it works, communicates out to Chinese servers, disguises itself as a Tire Pressure Monitor Sensor app, etc.

https://xdaforums.com/t/infinix-note-10-uninstall-tpms-com-hoffnung-package-causes-bootloop.4647456/

ThatGreenBlur, 2024-05-30:
> Is this on the Infinix Note 10? These people were able to get a decompiled version of the APK over company email after some resistance. There's more going on with how it works, communicates out to [...]

this is based on my own testing where simply just disabling the app completely breaks lock sreen notifications while reenabling immediately fixes it on my Infinix HOT 12 Play. I would still recommend this to be set to advanced.

Rudxain, 2024-09-15:
Half-fixed by #609. We still need to include the lock-screen problem in the des

AnonymousWP, 2024-09-15:
I'll work on adding packages today.