Universal-Debloater-Alliance/universal-android-debloater-next-generation #490: docs(wiki): explain differences between disabling and uninstalling

AnonymousWP, 2024-05-31:
### Describe the documentation you want to change

Users should be able to make a wise choice whether they want to disable or uninstall a package. We're planning to outline the benefits and downsides for each removal method.

### Acknowledgements

- [X] This issue is not a duplicate of an existing documentation change request.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Madis0, 2024-11-23:
[This Stackoverflow answer](https://android.stackexchange.com/a/64953) might help with that documentation, though I am unsure how up to date it is.

Rudxain, 2025-01-07:
#670 seems like useful info, too

Rudxain, 2025-01-23:
[Specific link](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/FAQ/9c2569eb9b6bf9067369b52ea79a699e9f5b6359#whats-the-difference-between-disabling-and-uninstalling)

closed 2025-01-23 by Rudxain