Universal-Debloater-Alliance/universal-android-debloater-next-generation #491: pkg(scope): miui.systemui.plugin

spaceXrace, 2024-05-31:
### List your packages

- Package name: miui.systemui.plugin
- Current list: OEM
  - Proposed list: OEM
- Current removal: Advanced
  - Proposed removal: Advanced
- Current description: System UI Plug-in. When removed, Android will use the AOSP style volume bar.
  - Proposed description: System UI Plug-in. When removed, Android will use the AOSP style volume bar and reboot menu. Will break the iOS-style quick settings.

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Blitz517, 2024-05-31:
what do you mean by reboot menu?

spaceXrace, 2024-05-31:
The menu that appears when you hold the power button, which gives you the option to turn off the device or reboot it. Maybe reboot menu wasn't the best description, sorry!

-------- Original Message --------

Blitz517, 2024-05-31:
its only on hyperos or beta miui? on my phone i only lose miui style volume bar

spaceXrace, 2024-05-31:
Interesting. I'm on HyperOS, I loose all the things mentioned.

-------- Original Message --------

nyveZz, 2024-06-06:
Can confirm on HyperOS. It also breaks the control center and reboot options for me.

spaceXrace, 2024-06-06:
Are your reboot options broken or just back to AOSP? Mine are AOSP, but I am on the chinese version

-------- Original Message --------

AnonymousWP, 2024-06-10:
I reflected this conversation in the linked PR.

closed 2024-06-27 by AnonymousWP

cameronj86, 2024-12-12:
Nitpicky, and yes I should've read the entire description before uninstalling instead of just skimming, but since losing system Quick Settings will have a larger impact to the average user than gaining AOSP volume setting, perhaps it can be listed first? I can do the PR myself if desired.

AnonymousWP, 2024-12-12:
> Nitpicky, and yes I should've read the entire description before uninstalling instead of just skimming, but since losing system Quick Settings will have a larger impact to the average user than gaining AOSP volume setting, perhaps it can be listed first? I can do the PR myself if desired.

Yes please open a PR and I'll have a look at it.