Universal-Debloater-Alliance/universal-android-debloater-next-generation #499: bug(scope): F-Droid Apps Not showing in uad-ng-linux

accforgithubtest, 2024-06-04:
### Describe the bug

All the apps installed via f-droid or side-loaded are not showing up in UAD-NG.
I have selected All (Lists) | All Packages | All Lists for user 0.

### Expected behavior

All Apps (including user installed apps via third-party app stores or side loaded) on the device should be listed.

### Provide logs

Pls let me know what logs are needed and how I can obtain the logs.

I am using uad-ng-linux  v1.0.3 version. Seeing this issue with Lenovo TB128FU tablet.
Any app installed from f-droid (OSMAnd, Hacki, etc, etc) do not show up in the apps list.

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-06-04:
I don't really understand what you expected. UAD-ng is only to remove bloatware, i.e. pre-installed applications. What applications do you mean? Can you give an example of a package name? UAD-ng is not meant to remove all kinds of applications. We have a curated list of applications: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/main/resources/assets/uad_lists.json.

Logs: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/FAQ#where-are-the-logfiles

accforgithubtest, 2024-06-11:
> UAD-ng is only to remove bloatware

Ah ok, I assumed UAD-ng can remove all installed applications along with the bloatware. 
I understand the need for the curated list of applications, to classify whats safe to remove and what not to remove, etc.

But can UAD-ng allow all user installed applications to be removed, as a new feature?

AnonymousWP, 2024-06-23:
You can file a feature request for that. 🙂 We can't guarantee we'll implement it, but then it'll be on the backlog at least.

accforgithubtest, 2024-06-24:
New feature request - #515 515

closed 2024-06-24 by accforgithubtest