Universal-Debloater-Alliance/universal-android-debloater-next-generation #501: bug(compilation): Error compiling from source

Minoxs, 2024-06-08:
### Describe the bug

When compiling from source (with or without --release) an error is raised about ambiguous naming in the iced_style package.

[image]

The next one might also happen:

[image]

### Expected behavior

UAD compiles without error.

### You have a solution?

Perhaps not the correct solution, but as a temporary workaround:

Look for the files which raised errors and change the following:

[image]

[image]

### Provide logs

N/A

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

B0ney, 2024-06-15:
Does `rustup update` solve the issue?

Minoxs, 2024-06-15:
I just tried it and it did not solve. Same issue.

I even cleared `.cargo\registry` after updating and still nothing.

Frigyes06, 2024-11-26:
Is this still an issue?

Minoxs, 2024-12-06:
Yep, still happening:

[image]

[image]

Cargo freshly updated. Windows 64 bits.

AnonymousWP, 2024-12-14:
> Yep, still happening: [...]

Would you be able to fix this and open a PR? Would be nice.

Minoxs, 2024-12-14:
I can definitely try! I opened an issue here because I'm unsure if this only happens to me or if it happens to others as well. It would be nice if someone else could try and confirm if this happens on another machine as well.

Minoxs, 2024-12-15:
I fiddled with this for a bit and realised I had the wrong toolchain active (version 1.70). After going to stable and updating to 1.83 it compiles just fine.

My bad!!

closed 2024-12-15 by Minoxs