Universal-Debloater-Alliance/universal-android-debloater-next-generation #507: User 0 uninstall --> user 999 restore (vice versa)

wezfaas, 2024-06-13:
### Describe the bug

I have second space (user 999) enabled only for whatsapp.  Somehow if I uninstall an app from user 0, UAD decides to restore it on user 999.  It sorts of render the tools unusable, ecause it will create a "new copy" for user 999, which shouldn't exist in the first place.

### Expected behavior

User 0 uninstall --> user 999 restore
User 999 uninstall --> user 0 restore

### Provide logs

1 package selected, 2 operations!

[image]
[image]

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2025-01-09:
This is very likely a device-specific (OEM?) problem. Can you please share the log-file or test ADB commands? If it happens with ADB alone, then it's unlikely something we can fix :(

AnonymousWP, 2025-01-13:
@wezfaas

cristianastorino, 2026-08-25:
I see the same behaviour with latest stable version 1.2.0. Looks like this problem is not fixed