Universal-Debloater-Alliance/universal-android-debloater-next-generation #508: feat(scope): Download ADB automatically on the first run

maverick74, 2024-06-19:
### Describe the feature you want

I don't know if this is possible but i think it would be useful - specially for non-techies - if:

- **After installing UAD, on the first run, it would automatically download and install/extract ADB.**

This would simplify all the process to get UAD working and reduce it to a **_kind of_** "_one installation process and you're ready to go_"!

To a user the experience would be something like:

   1. Install UAD,
   2. connect the phone to PC and
   3. remove the apps you want

(don't know if this is possible, however... just an idea ;). Feel free to close it as nonsense if it feels like it! )

### Acknowledgements

- [X] This issue is not a duplicate of an existing feature request.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2024-09-22:
An alternative to this is to embed ADB within UAD-ng itself. There are 2 main ways to do so:
- AppImage (#577)
- [A Crate](https://github.com/cocool97/adb_client)

I'd prefer the crate, as it would allow:
- us to choose an exact ADB version (more stable and predictable than an auto-updatable ADB)
- the compiler to further optimize and shrink the sizes of UAD+ADB

maverick74, 2024-09-22:
> An alternative to this is to embed ADB within UAD-ng itself. There are 2 main ways to do so:
> 
>     * AppImage ([Considering other distribution methods #577](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/discussions/577))

Considering Linux I, personally, would prefer AppImage!