Universal-Debloater-Alliance/universal-android-debloater-next-generation #509: bug(scope): no decides/emulators found

vitormarchini, 2024-06-20:
### Describe the bug

I've been able to use the UAD other times normally. Today I tried to use it normally, but it doesn't identify my device in any way, even though I leave USB debugging active. What happens?

### Expected behavior

I want my device to be recognized again so I can delete the app I want.

### Provide logs

[image]

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-06-23:
Please [post logs](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/wiki/FAQ#where-are-the-logfiles).

edgy-b, 2024-07-04:
@vitormarchini 
Go to Developer and try 
 - revoking USB connections
 - turn on/off USB debugging

AnonymousWP, 2024-09-10:
Closing this as we gave a solution for the problem, and also because we didn't get any reply.

closed 2024-09-10 by AnonymousWP