Universal-Debloater-Alliance/universal-android-debloater-next-generation #510: pkg(scope): Google play system update

WPHorhe, 2024-06-20:
### Describe the problem

Your phone model
Google Pixel 7a

Describe the issue
I used ShizuTools app to debloat my phone everything is nice and easy but I disabled something and since then when I go into device settings and I go to check for Google play system update I press on it and it tries to open the menu but it brings me back trying multiple times doesn't work I even disabled Google play store and enabled , clear data .. but it doesn't work I have this issue since I used shizu tools to debloat my phone about a month ago.
I was just asking if u can provide a package name so I cann go back and re-enable it since I disable so much apps and I don't know which may cause the issue .

Also I disabled an package named (Main Components , package name : com.google.mainline.adservices) marked as recommend to uninstall , which kicked me to the home screen and disabled wifi debugging and I had no developer options in the settings I just performed a restart and I got developer options back.

### You have a solution?

No , I am asking for a solution

### Provide logs

I don't have

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-06-23:
Main Components is related to Google Play System update. Try re-enabling it. So that's `com.google.mainline.adservices`. Let us know if that fixed it and we'll adjust the description/removal accordingly.

WPHorhe, 2024-06-23:
i reenabled everything a few days ago and debloated again because i said maybe i disabled something by mistake , the service 
com.google.mainline.adservices is already enabled on my device because i told you in the post that disableing that service disables me the developer options tab and i get it back only by performing a restart of the device (restart not reset) even tho its not researched for every manufaturer and its marked safe to remove

AnonymousWP, 2024-06-27:
So what do you exactly expect us to do now?

closed 2024-06-27 by WPHorhe