Universal-Debloater-Alliance/universal-android-debloater-next-generation #511: pkg(scope): Samsung Dex gone?

kajdo, 2024-06-22:
### Describe the problem

I wanted to debloat my samsung galaxy tab s7 fe -- and I think i debloated more than expected

I loaded the recommended list, manually disabled everything i found related to notes and handwriting and started the process

I just tried to connect a external display and realized that dex is not starting up -- only screen mirror like standard android

I tried to start dex on device (without any display connected) -- doesnt start

Im afraid the recommended list removes dex

### You have a solution?

no idea -- tried to sideload the apk but wasnt successful

### Provide logs

i unpacked the linux download and run unbloater ... there is no log found in the described directory

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-06-23:
Did you make an export of what you deleted/uninstalled/disabled? If so, please post it.

kajdo, 2024-06-23:
> Did you make an export of what you deleted/uninstalled/disabled? If so, please post it. Also try re-enabling/installing the packages.

unfortunately not ... used it the first time and wasn't sure how its working

its not a big deal i coud reset it to default and redo trying to figure out whats the dex packages i should keep.

i guess there is no restore all thingy - right (currently not near my machine to check)

kajdo, 2024-06-23:
*FIX:*
I reconnected my S7FE, chose "uninstalled" marked everything and restored it with the UAD function

rebooted device

Dex is back and working

..... I still think, that there should be a review of the "recommend" packages, because of the description in the wiki i would not expect dex to break (but it's of course arguable if it's a negative package)
> Pointless or outright negative packages, and/or apps available through Google Play.

I would suggest to put it at least into "advanced"
>  Breaks obscure or minor parts of functionality

I did not yet find the package responsable for the issue - but i will play around and try to figure out which package causes dex to break and report it in a "feature suggestion" - I think this issue can be closed but wanated to leave a commend if another user faces the same issue + the description how to make it work again

great tool altogether - thx for making it easy to debloat in such an elegant way :)

AnonymousWP, 2024-06-27:
Good to hear your issue got solved. 🙂 Let us know if you figured out which package was responsible and we'll change its description and removal accordingly.

kajdo, 2024-12-18:
sorry ... tbh, i forgot - but will do it during the holidays and report back

kajdo, 2025-01-09:
Hi,

I finally managed to do the analysis ... here we go

I would suggest to take out following packages from the *recommended* list for the galaxy s7fe (SM-T733):
[image]

I did not intensively test if the handwriting sdk can be removed, but since I use noteshelf quite often i don't want to risk that even if it works directly after removing, that I experience some issues later on

klmsagent and the smdms fail during removal and throw an error which could lead to inconsistent removal of packages because if removed in bulk, and in the middle there is an issue, the other selected ones might be affected (not traced down 100% but some inconsistencies appeared during my tests -- see my "casual protocol" attached)

the others break DEX

Because I guess for some, DEX and handwriting are some Features which made them buy the tab (for example for me) - I would not put those packages on a "recommended to remove" list which in wiki is descirbed as  "Pointless or outright negative packages, and/or apps available through Google Play."

sorry for taking that long - thx for the software you provide, but please cosnider my input ... not everybody might be ok with the outcome i think 

[DEX - Test Debloater.pdf](https://github.com/user-attachments/files/18368218/DEX.-.Test.Debloater.pdf)

AnonymousWP, 2025-02-04:
> Hi,
>
> I finally managed to do the analysis ... here we go
>
> I would suggest to take out following packages from the _recommended_ list for the galaxy s7fe (SM-T733): [...]

Hi, sorry for getting back to you late. Could you please open a new issue with your recommendations/suggestions? Then I can work on it.

kajdo, 2025-03-12:
as requested, I created a new issue ... will close this one in favour of #899

closed 2025-03-12 by kajdo