Universal-Debloater-Alliance/universal-android-debloater-next-generation #513: pkg(scope): Unable to remove Samsung Game Optimizing Service

Marekk2k, 2024-06-23:
### Describe the problem

It is not possible to fully Samsung Game Optimization Service. After clicking the "Uninstall" button, it is removed for a while, but then it comes back again after refreshing the list in UAD. 

My device: Galaxy A54 5G
Android Version: 14
[image]

### Provide logs

[UAD_20240623.log](https://github.com/user-attachments/files/15945027/UAD_20240623.log)

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2024-06-24:
[Cross-repo dupe](https://github.com/0x192/universal-android-debloater/issues/193)

Rudxain, 2024-06-24:
For the record, I want to add that the pack is non-removable from my SM-A315G (Android 12.0, One UI 4.1)

AnonymousWP, 2024-06-26:
What if you try disabling? Probably some other package is responsible for re-installing it. It likely monitors other apps such as this one, whether they exist or not.

Nono-Song, 2024-09-17:
I found this [forum](https://xdaforums.com/t/impossible-to-debloat-uninstall-game-optimizing-service.4061417/) discussion about this issue and possible reason behind the scene.
Anyway, I tried to disable instead of uninstalling, and it seems to work at a glance…

AnonymousWP, 2024-09-23:
> Add the suggestion to the GOS pack description. This is within-scope and trivial to implement, but it's "manual" for the users.

I agree, we could add that you can only disable it. I'll assign this one to my name. It doesn't require manual intervention; users just need to use the disable mode in UAD-ng.