Universal-Debloater-Alliance/universal-android-debloater-next-generation #514: bug: Shell cannot change component state for null to 1 when changing some packages

TruffelNL, 2024-06-23:
### Describe the bug

I have problems with activating secure folder on my Samsung Galaxy S22+, so I tried to enable some of the packages to see if that would solve the problem. When doing so (for instance on the `com.samsung.cmfa.AuthTouch` package, it shows the `Shell cannot change component state for null to 1` error on the command line. Some of the packages do work to enable them again.

### Expected behavior

I expected the package to be enabled again.

### Provide logs

```
2024-06-23 20:16:35 INFO  [src/gui/views/list.rs:145] ANDROID_SDK: 34 | DEVICE: samsung SM-S906B
2024-06-23 20:16:35 INFO  [src/gui/views/list.rs:149] -----------------------------------------------------------------
2024-06-23 20:17:14 ERROR [src/core/sync.rs:129] [Advanced] pm enable --user 0 com.samsung.cmfa.AuthTouch -> 
Exception occurred while executing 'enable':
java.lang.SecurityException: Shell cannot change component state for null to 1
	at com.android.server.pm.PackageManagerService.setEnabledSettings(PackageManagerService.java:4360)
	at com.android.server.pm.PackageManagerService.-$$Nest$msetEnabledSettings(PackageManagerService.java:0)
	at com.android.server.pm.PackageManagerService$IPackageManagerImpl.setApplicationEnabledSetting(PackageManagerService.java:6424)
	at com.android.server.pm.PackageManagerShellCommand.runSetEnabledSetting(PackageManagerShellCommand.java:2537)
	at com.android.server.pm.PackageManagerShellCommand.onCommand(PackageManagerShellCommand.java:282)
	at com.android.modules.utils.BasicShellCommandHandler.exec(BasicShellCommandHandler.java:97)
	at android.os.ShellCommand.exec(ShellCommand.java:38)
	at com.android.server.pm.PackageManagerService$IPackageManagerImpl.onShellCommand(PackageManagerService.java:7084)
	at android.os.Binder.shellCommand(Binder.java:1104)
	at android.os.Binder.onTransact(Binder.java:914)
	at android.content.pm.IPackageManager$Stub.onTransact(IPackageManager.java:4905)
	at com.android.server.pm.PackageManagerService$IPackageManagerImpl.onTransact(PackageManagerService.java:7068)
	at android.os.Binder.execTransactInternal(Binder.java:1380)
	at android.os.Binder.execTransact(Binder.java:1311)
```

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2024-09-07:
This is not a bug in UAD-ng itself. But (as suggested by [`8itlew7r`](https://github.com/8itlew7r)) we could add a workaround cmd as fall-back for when this error happens, so that users don't need to run it manually

AnonymousWP, 2024-09-11:
> This is not a bug in UAD-ng itself. But (as suggested by [`8itlew7r`](https://github.com/8itlew7r)) we could add a workaround cmd as fall-back for when this error happens, so that users don't need to run it manually

Is this hard to implement, or does it have priority in your opinion?

Rudxain, 2024-09-13:
> Is this hard to implement

A "hacky" (hot)fix seems easy, such as running the fall-back cmd whenever there's any error.

A proper fix would require reading some docs, as the error may be different depending on Android version.

> or does it have priority in your opinion?

I guess this is "mid priority"? ("mid" relative to what?). It's definitely not critical (unless UAD-ng `panic`s because of this error)

Rudxain, 2024-09-16:
[Found more info](https://stackoverflow.com/questions/67954958/why-do-some-androids-show-permission-denied-while-some-dont-both-are-not-rooted/68148926#68148926)

Rudxain, 2024-09-16:
As I suspected, the error message may be different, depending on device.

Since it's non-standard behavior, ADB documentation won't help here. So our only choice is to guess ("hacky" fix)

AnonymousWP, 2025-03-25:
I just faced the issue myself as well.

Cosm-gedenke, 2025-12-30:
I'll test it in a few VMs I have with android to confirm, in any case it does seem to be an android issue, the fallback should be simple, as @Rudxain said. I'll look into it

Cosm-gedenke, 2025-12-31:
Ok so I have tried running this on these VMs, specifically the Samsumg Galaxy S22+ image for android Studio. I have not found the package you gave in the description probably because the image probably doesn't have it. However after reading a bit more I am pretty sure it is an android issue. I will self assign this issue and try to solve it by the end of today.