Universal-Debloater-Alliance/universal-android-debloater-next-generation #516: feat(list-format): Migrate to RON

Rudxain, 2024-06-24:
### Describe the feature you want

I believe [`uad_lists`](https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/ee4db0c62bdc9379aa0b3d6b0c5ca4dd0bb8836e/resources/assets/uad_lists.json) may benefit from ["Rusty" Object Notation](https://github.com/ron-rs/ron)

AFAIK, there are tools for converting JSON to RON pretty easily

### Acknowledgements

- [X] This issue is not a duplicate of an existing feature request.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-06-26:
What are the benefits?

Rudxain, 2024-06-28:
> What are the benefits?

I was thinking of automatic self-validation of fields, such as `list` and `removal`, which can be defined as `enum`s.

However, we could also do that if we define our own JSON schema and use a schema validator. Then, we should setup a GH workflow/action to auto-run the validator on-push to `main` (applies to PRs).

I guess RON isn't really necessary, since the schema would be pretty simple to define. But this simplicity would also allow us to migrate faster

AnonymousWP, 2024-07-04:
But what current issue does that solve actually? Because we have GitHub Actions/Workflows for that that validate the JSON. Don't get me wrong; I appreciate your effort, but this project costs quite some time and it must be effective enough to work on.

Rudxain, 2024-07-04:
I didn't know the JSON was validated by workflows 😅. I'll close now lmao

closed 2024-07-04 by Rudxain

Rudxain, 2024-09-06:
[RON supports multi-line strs](https://github.com/ron-rs/ron/pull/112) (TOML does, too), while [JSON doesn't](https://stackoverflow.com/questions/2392766/are-multi-line-strings-allowed-in-json). This would help a lot when writing descriptions