Universal-Debloater-Alliance/universal-android-debloater-next-generation #517: pkg(G Play Services): Sync broken, Gmail cannot detect G account

Rudxain, 2024-06-24:
### Describe the problem

After uninstalling and reinstaling `com.google.android.gms`, the "Contacts" option in Google account sync no longer shows up, and "Google One" no longer backs-up Contacts, forcing me to do it manually with vCard files.

After reinstalling `com.google.android.gm` (Gmail, not G Play Services), it no longer detects **any** Google account, even if it shows up in the system's account list and it was added post-GMS-reinstall.

I opened this issue because this should be documented, not because I believe UAD is wrong. I knew very well that uninstalling GMS would break stuff, but I reinstalled it because it broke too much stuff and [microG](https://github.com/microg/GmsCore) requires unlocked bootloader.

- Android: 12.0
- Model: SM-A315G

### You have a solution?

I've tried:
- clearing GMS data
- removing and re-adding my account
- rebooting after doing either of above

Unfortunately, the only known reliable solution is a factory reset

### Provide logs

`None`

### Acknowledgements

- [X] This issue is not a duplicate of an existing one.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2024-08-15:
Setup Wizard is uninstalled, I'm suspecting that's why Accounts don't work properly

Rudxain, 2024-08-27:
Gmail now detects account after enabling SetupWizard!

Still can't sync Contacts. Guess, I'll have to factory-reset

AnonymousWP, 2024-08-27:
Good to know, thanks for confirming.

AnonymousWP, 2024-09-18:
@Rudxain Quick question. So do you want me to add this to the description of which package? There are many packages with ` SetupWizard` in the name.

Rudxain, 2024-09-18:
`com.google.android.setupwizard`. It makes sense considering the "mobile identity management"

Rudxain, 2024-09-19:
Should I change the title? The real culprit was SW, not GMS. But I still don't know what happened to Calendar and Contacts

closed 2024-09-19 by AnonymousWP

Rudxain, 2024-11-28:
The Calendar magically works again! It seems Google has updated Play Services this week, Now S Calendar properly fetches all the events from all my accounts!

Contact-sync seems to have been working for a longer time, but IDK why. Maybe I did something?