Universal-Debloater-Alliance/universal-android-debloater-next-generation #519: bug(ci): lint fails

AnonymousWP, 2024-06-24:
### Describe the bug

With PRs, the linting step is failing: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/actions/runs/9650187284/job/26615292076#step:6:718.

### Expected behavior

I expect the checks to pass.

### Provide logs

CI / GitHub Action logs: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/actions/runs/9650187284/job/26615292076#step:6:718

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

adhirajsinghchauhan, 2024-06-24:
Unused var is rarely a useful lint failure. I propose to disable that if possible. Most of the time it's only an annoyance.

However, does anyone know why `clippy` never reported this failure earlier? It's odd that a linter is flaky.

AnonymousWP, 2024-06-26:
Yeah that's odd. Anyway, seems like it's a warning anyway, but still fails. Can you confirm that when I remove scanning for warnings (https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/blob/main/.github/workflows/ci.yml#L41), this should solve the issue?

adhirajsinghchauhan, 2024-06-29:
Removing `-- -D warnings` will still report warnings in logs, but it won't fail the build. This flag only serves to set the error code to a non-zero value.

AnonymousWP, 2024-06-29:
> Removing `-- -D warnings` will still report warnings in logs, but it won't fail the build. This flag only serves to set the error code to a non-zero value.

Any idea how to fix this? Isn't that what we want? We want to be informed about warnings, but that shouldn't make the build fail.

Rudxain, 2024-07-05:
Does [this](https://stackoverflow.com/questions/62045967/is-there-a-way-to-continue-on-error-while-still-getting-correct-feedback) apply in our case?

adhirajsinghchauhan, 2024-07-05:
What I said in https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/519#issuecomment-2198124323 is how you resolve this issue. Sorry if I wasn't clear earlier.

PR #582 (2024-08-19): build(ci): treat warns as warns
fixes #519

https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/519#issuecomment-2198124323
closed 2024-08-19 by adhirajsinghchauhan