Universal-Debloater-Alliance/universal-android-debloater-next-generation #520: bug(device-malformed-packages-list): Some devices can output a malformed package list

mateusjdev, 2024-06-25:
### Describe the bug

This is not a bug on the app itself but can be useful as there issues like this https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/316 that i think are related. Using the app on my Redmi Note 3, no packages are shown, after testing, I discovered that the way the app are pulling packages from adb in this device is returning with a "\r" character after package names, causing the app to mismatch "com.app" and "com.app\r" . So when I do a backup of packages list on the app on the Redmi Note 3, the json is exported as follow:

```json
{
  "device_id": "c8343074",
  "users": [
    {
      "id": 0,
      "packages": [
        {
          "name": "android\r",
          "state": "Enabled"
        },
        {
          "name": "android.autoinstalls.config.Xiaomi.kate\r",
          "state": "Enabled"
        },
        {
          "name": "com.android.backupconfirm\r",
          "state": "Enabled"
        },
        {
          "name": "com.android.backupreceiver\r",
          "state": "Enabled"
        },
        {
          "name": "com.android.bluetooth\r",
          "state": "Enabled"
        },
        {
          "name": "com.android.bluetoothmidiservice\r",
          "state": "Enabled"
        }
      ]
    }
  ]
}
```
All of this cause a strange behavior: no packages are shown in recommended list but many packages on device are in the uad_list.json as shown in the screenshot:

Redmi Note 3 (Recommended) | Redmi Note 3 (All Packages + Search)
Redmi Note 3 (adb shell pm list packages) | Xiaomi Mi 9 (Recommended + Search)

[image]

As in the screenshot, the Xiaomi Mi 9 and other devices i tested work correctly.

### Expected behavior

To packages like "com.app\r" show on the same lists and description as "com.app".

### You have a solution?

Try to sanitize package names.

### Provide logs

2024-06-25 17:49:22 DEBUG [src\core\update.rs:174] Checking for UAD update
2024-06-25 17:49:24 INFO  [src\gui\views\list.rs:125] -----------------------------------------------------------------
2024-06-25 17:49:24 INFO  [src\gui\views\list.rs:126] ANDROID_SDK: 23 | DEVICE: Xiaomi Redmi Note 3
2024-06-25 17:49:24 INFO  [src\gui\views\list.rs:130] -----------------------------------------------------------------
2024-06-25 17:49:31 INFO  [src\gui\mod.rs:248] -----------------------------------------------------------------
2024-06-25 17:49:31 INFO  [src\gui\mod.rs:249] ANDROID_SDK: 23 | DEVICE: Xiaomi Redmi Note 3
2024-06-25 17:49:31 INFO  [src\gui\mod.rs:253] -----------------------------------------------------------------
2024-06-25 17:49:34 INFO  [src\gui\views\settings.rs:133] [BACKUP] Backup successfully created
2024-06-25 17:49:36 ERROR [src\core\utils.rs:83] Can't open the following URL: 
2024-06-25 17:53:01 INFO  [src\gui\views\list.rs:125] -----------------------------------------------------------------
2024-06-25 17:53:01 INFO  [src\gui\views\list.rs:126] ANDROID_SDK: 0 | DEVICE: fetching devices...
[... 80 lines cut ...]
2024-06-25 18:11:10 INFO  [src\gui\views\list.rs:144] -----------------------------------------------------------------
2024-06-25 18:11:10 INFO  [src\gui\views\list.rs:145] ANDROID_SDK: 23 | DEVICE: Xiaomi Redmi Note 3
2024-06-25 18:11:10 INFO  [src\gui\views\list.rs:149] -----------------------------------------------------------------
2024-06-25 18:11:25 DEBUG [src\core\update.rs:180] Checking for UAD-ng update
2024-06-25 18:11:27 INFO  [src\gui\views\list.rs:144] -----------------------------------------------------------------
2024-06-25 18:11:27 INFO  [src\gui\views\list.rs:145] ANDROID_SDK: 23 | DEVICE: Xiaomi Redmi Note 3
2024-06-25 18:11:27 INFO  [src\gui\views\list.rs:149] -----------------------------------------------------------------
2024-06-25 18:13:20 INFO  [src\gui\views\list.rs:144] -----------------------------------------------------------------
2024-06-25 18:13:20 INFO  [src\gui\views\list.rs:145] ANDROID_SDK: 23 | DEVICE: Xiaomi Redmi Note 3
2024-06-25 18:13:20 INFO  [src\gui\views\list.rs:149] -----------------------------------------------------------------
2024-06-25 18:13:36 INFO  [src\gui\views\list.rs:144] -----------------------------------------------------------------
2024-06-25 18:13:36 INFO  [src\gui\views\list.rs:145] ANDROID_SDK: 23 | DEVICE: Xiaomi Redmi Note 3
2024-06-25 18:13:36 INFO  [src\gui\views\list.rs:149] -----------------------------------------------------------------
2024-06-25 18:13:51 DEBUG [src\core\update.rs:180] Checking for UAD-ng update
2024-06-25 18:13:53 INFO  [src\gui\views\list.rs:144] -----------------------------------------------------------------
2024-06-25 18:13:53 INFO  [src\gui\views\list.rs:145] ANDROID_SDK: 30 | DEVICE: Xiaomi MI 9
2024-06-25 18:13:53 INFO  [src\gui\views\list.rs:149] -----------------------------------------------------------------
2024-06-25 18:29:02 DEBUG [src\core\update.rs:180] Checking for UAD-ng update
2024-06-25 18:29:05 ERROR [src\core\utils.rs:78] Can't open the following URL: 

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

mateusjdev, 2024-06-25:
To make sure I was not being dumb and as I saw some other bug reports about system vs user packages, I tested on linux, which surprisingly showed all recommended packages normally (and no "\r" character in the backup file):

[image]

AnonymousWP, 2024-07-04:
@Universal-Debloater-Alliance/developers Not sure if it's an easy fix, nor do I know if this impacts a lot of users.

Rudxain, 2024-07-04:
If this only happens with Xiaomi devices connected to Windows, I'd be pretty surprised! If that's the case, then the drivers may be to blame. Does your Windows device have some special Xiaomi software installed (other than bare drivers)? Some GUI programs come bundled with drivers, such as Intel Graphics settings

mateusjdev, 2024-07-04:
> Does your Windows device have some special Xiaomi software installed (other than bare drivers)?

None, only adb, I actually tried version 34 and 35 but they do the same.

> Not sure if it's an easy fix, nor do I know if this impacts a lot of users.

Yeah, thinking about it, as it work on linux it's not a big problem for me, and as it a 8 year old device I think not many others have this problem. Think I can close this.

closed 2024-07-04 by mateusjdev

Rudxain, 2024-07-05:
If anyone reading this manages to repro with non-Xiaomi devices, please let us know