Universal-Debloater-Alliance/universal-android-debloater-next-generation #523: bug(scope): Can't read anything from the latest release v1.0.3.

ramesson, 2024-06-27:
### Describe the bug

Can't read anything from the latest release v1.0.3.
v1.0.2 works fine 
OS: PopOS on X11
pic attached
[image]

### Expected behavior

to work like v1.0.2 visually.

### Provide logs

[UAD_20240627.log](https://github.com/user-attachments/files/16017824/UAD_20240627.log)

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-07-04:
Not sure if it's related to an issue another PopOS user had, but try running it using `sudo`. Not sure if it's related though.

DenizThatMenace, 2024-07-14:
I have the same problem.
`git bisect`ing points to the following commit as the culprit: f643f1dacf0a06b3cd03ecb400879290553a1def

Rudxain, 2024-07-16:
This is certainly a font/typeface problem. The weird thing is that "Search packages" renders properly. If `iced` depends on `wgpu` then I suspect it may be related to `wgpu`

AnonymousWP, 2024-09-15:
Closing this due to it being stale. Try using v.1.1.0.

Rudxain, 2024-09-22:
#641

kubijo, 2024-11-05:
I started creating new issue because of this same issue ... tried the last version on GH

[image]

## Warnings preset upon running the app: 

```
2024-11-05 14:40:51 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/sctk-adwaita-0.8.3/src/buttons.rs:172] Ignoring unknown button type: 
2024-11-05 14:40:51 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/gles/egl.rs:268] No config found!
2024-11-05 14:40:51 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/gles/egl.rs:257] EGL says it can present to the window but not natively
```

- Ubuntu 24.10 (Oracular Oriole)
- Wayland 1.23, GNOME Shell 47.0

<details><summary>Log</summary>

```
2024-11-05 14:40:50 ERROR [src/core/config.rs:94] Failed to read config file: `No such file or directory (os error 2)`
2024-11-05 14:40:50 ERROR [src/core/config.rs:96] Restoring default config file
2024-11-05 14:40:51 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/sctk-adwaita-0.8.3/src/buttons.rs:172] Ignoring unknown button type:
2024-11-05 14:40:51 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/gles/egl.rs:268] No config found!
2024-11-05 14:40:51 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/gles/egl.rs:257] EGL says it can present to the window but not natively
2024-11-05 14:40:51 DEBUG [src/core/update.rs:183] Checking for UAD-ng update
2024-11-05 14:41:51 INFO  [src/gui/views/list.rs:160] -----------------------------------------------------------------
2024-11-05 14:41:51 INFO  [src/gui/views/list.rs:161] ANDROID_SDK: 0 | DEVICE: fetching devices...
2024-11-05 14:41:51 INFO  [src/gui/views/list.rs:165] -----------------------------------------------------------------
2024-11-05 14:41:52 ERROR [src/gui/views/list.rs:873] AppsView ready but no phone found
2024-11-05 14:47:08 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/sctk-adwaita-0.8.3/src/buttons.rs:172] Ignoring unknown button type:
2024-11-05 14:47:08 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/gles/egl.rs:268] No config found!
2024-11-05 14:47:08 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/gles/egl.rs:257] EGL says it can present to the window but not natively
2024-11-05 14:47:08 DEBUG [src/core/update.rs:183] Checking for UAD-ng update
2024-11-05 14:47:17 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/sctk-adwaita-0.8.3/src/buttons.rs:172] Ignoring unknown button type:
2024-11-05 14:47:17 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/gles/egl.rs:268] No config found!
2024-11-05 14:47:17 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/gles/egl.rs:257] EGL says it can present to the window but not natively
2024-11-05 14:47:18 DEBUG [src/core/update.rs:183] Checking for UAD-ng update
```
</details>

AnonymousWP, 2025-01-13:
@ramesson @kubijo @DenizThatMenace Sorry that it took this long. Could you try with one of these artifacts? https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/actions/runs/12749084042

kubijo, 2025-01-15:
> @ramesson @kubijo @DenizThatMenace Sorry that it took this long. Could you try with one of these artifacts? https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/actions/runs/12749084042

[image]

AnonymousWP, 2025-01-15:
> ![image](https://private-user-images.githubusercontent.com/11244314/403339610-4f7881cc-788c-4791-9529-5d06737c39ee.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Could you please post some logs?

kubijo, 2025-01-15:
```log
2025-01-15 10:59:44 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/sctk-adwaita-0.8.3/src/buttons.rs:172] Ignoring unknown button type: 
2025-01-15 10:59:44 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/gles/egl.rs:268] No config found!
2025-01-15 10:59:44 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/gles/egl.rs:257] EGL says it can present to the window but not natively
2025-01-15 10:59:45 DEBUG [src/core/update.rs:183] Checking for UAD-ng update
2025-01-15 11:00:18 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/sctk-adwaita-0.8.3/src/buttons.rs:172] Ignoring unknown button type: 
2025-01-15 11:00:18 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/gles/egl.rs:268] No config found!
2025-01-15 11:00:18 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/gles/egl.rs:257] EGL says it can present to the window but not natively
2025-01-15 11:00:19 DEBUG [src/core/update.rs:183] Checking for UAD-ng update
2025-01-15 11:00:40 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/sctk-adwaita-0.8.3/src/buttons.rs:172] Ignoring unknown button type: 
2025-01-15 11:00:40 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/gles/egl.rs:268] No config found!
2025-01-15 11:00:40 WARN  [/home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.4/src/gles/egl.rs:257] EGL says it can present to the window but not natively
2025-01-15 11:00:40 DEBUG [src/core/update.rs:183] Checking for UAD-ng update
```

kubijo, 2025-02-17:
Makes no difference 

[image]

AnonymousWP, 2025-02-19:
@kubijo This is apparently a different issue than the (now unlinked) PR, see this comment: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/pull/850#issuecomment-2667095823

PR #858 (2025-02-22): fix: use advanced text shaping to prevent text being rendered as squares with certain fonts or CJK characters
I was only able to reproduce #523 with CJK characters. I tried using default Pop!_OS fonts from their PPA, but they were rendering normally.

This fix is based on the following comments:
- https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/issues/523#issue-2378550768 the input field is rendered properly.
- https://github.com/iced-rs/iced/issues/2709 again, the input field is rendered properly, and the user notes "Adding shaping(text::Shaping::Ad [... cut]
closed 2025-02-22 by Rudxain