Universal-Debloater-Alliance/universal-android-debloater-next-generation #526: bug(scope): Unable to compile on Windows

clin1234, 2024-06-30:
### Describe the bug

Installing via `cargo install --git https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/`, encountered this error:

```
error: linking with `link.exe` failed: exit code: 1120
  |
  = note: "C:\\Program Files\\Microsoft Visual Studio\\2022\\Preview\\VC\\Tools\\MSVC\\14.41.33923\\bin\\HostX64\\x64\\link.exe" "/NOLOGO" "C:\\Users\\0BAC~1\\AppData\\Local\\Temp\\rustc2ep1Mv\\symbols.o" "C:\\Users\\0BAC~1\\AppData\\Local\\Temp\\cargo-installG53cfB\\release\\deps\\uad_ng.uad_ng.f300ab40b6aed1e3-cgu.08.rcgu.o" "/LIBPATH:C:\\Users\\0BAC~1\\AppData\\Local\\Temp\\cargo-installG53cfB\\release\\deps" "/LIBPATH:C:\\Users\\ךינשגכהד\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\windows_x86_64_msvc-0.52.5\\lib" "/LIBPATH:C:\\Users\\ךינשגכהד\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\windows_x86_64_msvc-0.48.5\\lib" "/LIBPATH:C:\\Users\\0BAC~1\\AppData\\Local\\Temp\\cargo-installG53cfB\\release\\build\\ring-349ef5e2c0ffe087\\out" "/LIBPATH:C:\\Users\\ךינשגכהד\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib" "C:\\Users\\0BAC~1\\AppData\\Local\\Temp\\rustc2ep1Mv\\libring-d0982aaaf8a938bb.rlib" "C:\\Users\\0BAC~1\\AppData\\Local\\Temp\\rustc2ep1Mv\\libstd-cef76c2685dfb4ca.rlib" "C:\\Users\\ךינשגכהד\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib\\libcompiler_builtins-a6954cc40265b962.rlib" "windows.0.52.0.lib" "bcrypt.lib" "advapi32.lib" "kernel32.lib" "user32.lib" "shell32.lib" "gdi32.lib" "advapi32.lib" "windows.0.48.5.lib" "kernel32.lib" "ole32.lib" "opengl32.lib" "windows.0.52.0.lib" "windows.0.52.0.lib" "windows.0.52.0.lib" "windows.0.52.0.lib" "advapi32.lib" "cfgmgr32.lib" "d2d1.lib" "d3d12.lib" "d3dcompiler.lib" "dcomp.lib" "dwrite.lib" "dxgi.lib" "gdi32.lib" "kernel32.lib" "msimg32.lib" "ole32.lib" "oleaut32.lib" "opengl32.lib" "synchronization.lib" "user32.lib" "windowscodecs.lib" "winspool.lib" "windows.0.52.0.lib" "kernel32.lib" "advapi32.lib" "kernel32.lib" "ntdll.lib" "userenv.lib" "ws2_32.lib" "kernel32.lib" "ws2_32.lib" "kernel32.lib" "/defaultlib:msvcrt" "/NXCOMPAT" "/LIBPATH:C:\\Users\\ךינשגכהד\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib" "/OUT:C:\\Users\\0BAC~1\\AppData\\Local\\Temp\\cargo-installG53cfB\\release\\deps\\uad_ng.exe" "/SUBSYSTEM:windows" "/ENTRY:mainCRTStartup" "/OPT:REF,ICF" "/DEBUG" "/PDBALTPATH:%_PDB%" "/NATVIS:C:\\Users\\ךינשגכהד\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\etc\\intrinsic.natvis" "/NATVIS:C:\\Users\\ךינשגכהד\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\etc\\liballoc.natvis" "/NATVIS:C:\\Users\\ךינשגכהד\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\etc\\libcore.natvis" "/NATVIS:C:\\Users\\ךינשגכהד\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\etc\\libstd.natvis" "C:\\Users\\0BAC~1\\AppData\\Local\\Temp\\cargo-installG53cfB\\release\\build\\uad-ng-7a59dd021bb79a68\\out/manifest.lib"
  = note: symbols.o : error LNK2001: unresolved external symbol _ZN6uad_ng10CONFIG_DIR51__static_init_constructor___static_init_initializer17h5aa796223ac4b8a0E
          symbols.o : error LNK2001: unresolved external symbol _ZN6uad_ng4core4save10BACKUP_DIR51__static_init_constructor___static_init_initializer17h3e6ae80a7b3df659E
          symbols.o : error LNK2001: unresolved external symbol _ZN6uad_ng4core4sync13get_user_list2RE51__static_init_constructor___static_init_initializer17h7868eb56f6cf980aE
          symbols.o : error LNK2001: unresolved external symbol _ZN6uad_ng4core4sync2RE51__static_init_constructor___static_init_initializer17hbde9ce955a8301a1E
          symbols.o : error LNK2001: unresolved external symbol _ZN6uad_ng4core6config11CONFIG_FILE51__static_init_constructor___static_init_initializer17ha39b786cc017aef5E
          symbols.o : error LNK2001: unresolved external symbol _ZN6uad_ng9CACHE_DIR51__static_init_constructor___static_init_initializer17h9229706a25065ba5E
          C:\Users\0BAC~1\AppData\Local\Temp\cargo-installG53cfB\release\deps\uad_ng.exe : fatal error LNK1120: 6 unresolved externals
```

### Expected behavior

Successful compilation

### Provide logs

N/A

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

AnonymousWP, 2024-07-04:
I'm not sure as I'm not a Rust developer, but would you be able to help fixing this?

clin1234, 2024-07-04:
Huh. Updating my nightly toolchain solved this issue.

closed 2024-07-04 by clin1234