Universal-Debloater-Alliance/universal-android-debloater-next-generation #532: bug(settings): Panic if searching for devices

Rudxain, 2024-07-04:
### Describe the bug

0. `adb` not needed. In fact, it's easier to repro if it isn't available
1. Open UAD
2. Click the cog icon (settings) before a device is connected
3. Crash happens

### Expected behavior

Same as when UAD gives up on trying to find a device, which makes a non-existent "user 0" available, preventing the crash

### You have a solution?

2 alts:
- Correct: stop relying on an Android uid just to open settings
- Hacky: simply create the fake "user 0" before it's even detected

### Provide logs

thread 'main' panicked at src/gui/views/settings.rs:481:41:
called `Option::unwrap()` on a `None` value

### Acknowledgements

- [X] This issue is not a duplicate of an existing bug report.
- [X] I have chosen an appropriate title.
- [X] All requested information has been provided properly.

Rudxain, 2024-07-04:
I stumbled upon this when trying to debug #531, but it also happens on upstream.

It's annoying to have to connect a device just to debug faster. Maybe there should be a button or CLI flag to stop UAD from searching devices?

Rudxain, 2024-07-04:
I found the source of the regression: https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/pull/527/files#diff-ab8fdd0e30bade065fcdbadef147f14626783f8c82de0924f9312b80644edd36R481

It seems it'll be hard to avoid the "Hacky" solution I mentioned, unless all of the data dependencies can be moved to the conditional branch that actually needs them (I'm afraid there are useful side-effects that will be deferred). I'll give it a try...

Rudxain, 2024-07-04:
I found a more serious bug while comparing behaviors between upstream and https://github.com/Universal-Debloater-Alliance/universal-android-debloater-next-generation/pull/533/commits/f9501565276fc9809e7661c6108d172283cc3fe9:
1. Open UAD and wait for it to give up finding a device
2. Open settings
3. Sync (while Settings-view is still open)
4. (Panic happens)

AnonymousWP, 2024-07-04:
Good work on researching this!

> It's annoying to have to connect a device just to debug faster. Maybe there should be a button or CLI flag to stop UAD from searching devices?

Agreed.

Rudxain, 2024-07-04:
> Good work on researching this!

Thanks!

> Agreed.

I'll create an issue for that

PR #533 (2024-07-04): hotfix(settings): patch regression and refactor a bit
fixes #532
closed 2024-07-04 by edgy-b